Navideck/universal_ble #13: Scanning with services

martin-robert-fink, 2024-02-23:
I can't seem to find a way to scan for devices that only support a specific service (like an HR monitor).  It looks like I'd have to scan, connect, discover services, then disconnect if it doesn't support the services I want.  This doesn't seem like a very efficient way to do this.

Is there a way to scan and check services at the same time without having to connect to the device to get the list of services?

What am I missing?

Thanks,
Martin

fotiDim, 2024-02-26:
@martin-robert-fink which OS are you targeting? Does your HR monitor list the supported services in the advertisement data?

martin-robert-fink, 2024-02-26:
Hi -  Yes it does.  But, when I look at the [BleScanResult class](https://pub.dev/documentation/universal_ble/latest/universal_ble/BleScanResult-class.html), there is no list of ServiceUUIDs member.  So, how do I scan for BLE devices and filter on a specific serviceUUID so that only those devices (in my example HR Monitors) show up, or are returned in the list of discovered devices?

fotiDim, 2024-02-26:
We need to expose this functionality in Dart. Which OS are you targeting?

mortenboye, 2024-02-28:
I would love this for Windows as well.

fotiDim, 2024-03-05:
We have published 0.9.3 which adds the requested feature. Feel free to reopen if it does not work for you.

closed 2024-03-05 by fotiDim