Navideck/universal_ble #18: Windows: crash on start when no Bluetooth adapter exists

flo80, 2024-03-12:
In windows, when using the plugin (e.g. also example), the app crashes at startup when no Bluetooth adapter exists in the system. 

I seems to be a null pointer at some point in initialisation code.

fotiDim, 2024-03-13:
Thanks for the report. Fixed by https://github.com/Navideck/universal_ble/pull/20

Regular89, 2024-03-13:
Hi, 
The log issue on my side was:
√ Built build\windows\x64\runner\Debug\universal_ble_example.exe. Error connecting to the service protocol: failed to connect to http://127.0.0.1:50910/....
So thanks @fotiDim for the fix, it worked for me. 
 
However because there is not a new version, I had to apply the patch by hand instead of "flutter pub upgrade".
Any idea when the new version will be released ? 
I saw that I can reference the git repo directly but I would rather avoid this.

fotiDim, 2024-03-13:
We just released 0.9.4 which includes the fix

closed 2024-03-13 by fotiDim