Navideck/universal_ble #19: Disconnected peripherals disappear on Windows

BikeBr0, 2024-03-12:
I'm having trouble disconnecting from a BLE Peripheral. I am able to scan and connect to the peripheral but calling disconnect does not work as expected:

Steps to reproduce:

1) Use UniversalBle to scan and connect to a device on Windows
2) Confirmed with smartphome that device **does not** appear in bluetooth scans
3) Use UniversalBle to disconnect
4) Confirmed with smartphome that device **does** appear in bluetooth scans
5) Use UniversalBle to scan for devices

Result - BleDevice no longer available in UniversalBle  scan results
Expected - BleDevice should appear in scan results

rohitsangwan01, 2024-03-13:
@BikeBr0 After disconnected from UniversalBle, is that Ble peripheral also disconnected from Windows bluetooth settings, please cross check that
Also, verify if device is connected with: `UniversalBle.getConnectedDevices()` api

BikeBr0, 2024-03-13:
It must be @rohitsangwan01  since it is accessible from my phone. And I checked with getConnectedDevices, it was not returned.

rohitsangwan01, 2024-03-13:
@BikeBr0  ok so after disconnecting from UniversalBle
that devices appears in scan results from mobile
but not in windows ?

BikeBr0, 2024-03-13:
Correct

rohitsangwan01, 2024-03-13:
@BikeBr0 can you cross check with
https://apps.microsoft.com/detail/9n0ztkf1qd98?hl=en-US&gl=US
sometimes windows creates such issues

BikeBr0, 2024-03-13:
Hang on, this seems to work as expected with my very simple test case. I'll do more digging and let you know more precisely what is causing the problem (if I can).

BikeBr0, 2024-03-13:
All good @rohitsangwan01. I now see the bug in my own code.

closed 2024-03-13 by BikeBr0