Navideck/universal_ble #21: MSIX "release" package returns no scan results while "debug" package does

BikeBr0, 2024-03-14:
A strange error is occuring when I try to package my application into an MSIX installer for Windows using the flutter recommended msix plugin. I don't receive any errors when packaging and the app installs fine. However, when running the app, I don't receive any BleScanResults (but it doesn't throw an error). This same app running through VSCode or packaged as a debug build does work. Do you have any ideas why this is behaving this way? Sounds like a possible race condition if slower debug builds do work.

Performing the same test with a trivial app that only includes UniversalBle works fine as an MSIX package built in "release" mode.

rohitsangwan01, 2024-03-15:
@BikeBr0 example app seems to be working fine with MSIX, can you cross check with example app, if example app works for you then its hard to tell the exact issue, but you can share a minimal project which replicates this bug

BikeBr0, 2024-03-15:
It does work with my simple test app too @rohitsangwan01 . I've tracked it down to a behaviourly difference in the  BleScanResult stream. I am getting results but they are not appearing until the scanner is stopped (but this behaviour is limited to my actual app and not my simple test app). It's very strange as it does not occur in the DEBUG Build, only the RELEASE build. Joys of programming...

rohitsangwan01, 2024-03-15:
@BikeBr0 strange, again a test app which can replicate this behaviour will be helpful, also are you running something else with StartScan and StopScan methods, which might stop the ui thread, or do you notice any ui freeze while scan in progress ?

BikeBr0, 2024-03-15:
I think I have tracked down the bug @rohitsangwan01 . It seems the Company ID within the manufacturerData is not being set in the MSIX Release package. In my very simple test code, I only see [0,0,1] whereas my dev or debug build contains our actual Company ID.

fotiDim, 2024-03-16:
@BikeBr0 can you please try `flutter run -d windows --release` and see if it works? I am not sure the issue is msix related.

fotiDim, 2024-03-16:
@BikeBr0 I made a [PR](https://github.com/Navideck/universal_ble/pull/24) about it. Can you please try the code from the `fix-wrong-manufacturer-data-in-release-mode` branch and let me know if it works?

BikeBr0, 2024-03-16:
Will do. I'm away from my computer for a couple days but will check it as soon as I can.

BikeBr0, 2024-03-18:
> flutter run -d windows --release

I can confirm that this reproduces the bug

BikeBr0, 2024-03-18:
> @BikeBr0 I made a [PR](https://github.com/Navideck/universal_ble/pull/24) about it. Can you please try the code from the `fix-wrong-manufacturer-data-in-release-mode` branch and let me know if it works?

This has fixed the issue in my simple test app.

closed 2024-03-18 by BikeBr0