Navideck/universal_ble #26: Adding universal_ble to pubsepc.yaml results in "Nuget.exe not found" error/warning

BikeBr0, 2024-03-18:
I'm not sure if this is a problem, but when I add universal_ble as a dependency to pubspec.yaml (without actually using the plugin) in my simple hello_world app and run the app through VS Code, I see:

```
Launching lib\main.dart on Windows in debug mode...
Nuget.exe not found, trying to download or use cached version.
```

This occurs when running in release mode too:

```
C:\Users\me.person\workspace\hello_world>flutter run -d windows --release
Launching lib\main.dart on Windows in release mode...
Nuget.exe not found, trying to download or use cached version.
Building Windows application... 
```

I'm not sure if this is causing a runtime problem.

fotiDim, 2024-03-18:
I am pretty sure that it does not cause any runtime issues. I don't have nuget installed on my system and everything works fine. I have verified that with multiple computers. I am not sure if the plugin can suppress this warning somehow or if it just dependent on the development environment.

BikeBr0, 2024-03-18:
Sounds good @fotiDim. It's mainly worrisome since it prints out in red:

[image]

fotiDim, 2024-03-18:
[This](https://github.com/bluefireteam/audioplayers/pull/1449) is how other plugins are handling it.

fotiDim, 2024-03-20:
This is fixed in the latest release so closing.

closed 2024-03-20 by fotiDim

BikeBr0, 2024-03-20:
This does seem to be resolved @fotiDim . However, the console seems extra quiet now. I'm use to seeing more output while the application is being built. Now, the first output is from log statements in my own application code.

fotiDim, 2024-03-22:
@BikeBr0 the only thing we removed was this line. No other logs were removed. There are other logs that will print if something goes wrong.