Navideck/universal_ble #27: Calling startScan in widget's initState method throws PlatformException

BikeBr0, 2024-03-18:
This bug is specific to running on windows in release mode (cannot reproduce in debug mode). I can reliably reproduce this in a simple hello_world app but only occurs when startScan is called in the widget's initState method. Moving startScan to a button tap does not trigger the bug.

C:\Users\me.human\workspace\hello_world>flutter run -d windows --release
Launching lib\main.dart on Windows in release mode...
Nuget.exe not found, trying to download or use cached version.
Building Windows application...                                    19.4s
√  Built build\windows\x64\runner\Release\hello_world.exe (0.1MB).

Flutter run key commands.
h List all available interactive commands.
c Clear the screen
q Quit (terminate the application on the device).
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(Bluetooth is not available, , null, null)
#0      UniversalBlePlatformChannel.startScan (package:universal_ble/src/universal_ble_pigeon/universal_ble.g.dart:276)
<asynchronous suspension>
#1      UniversalBlePigeonChannel.startScan (package:universal_ble/src/universal_ble_pigeon/universal_ble_pigeon_channel.dart:35)
<asynchronous suspension>
#2      _QueuedFuture.execute (package:universal_ble/src/ble_command_queue.dart:66)
<asynchronous suspension>

BikeBr0, 2024-03-18:
Of note, when the Exception occurs, a subsequent tap to trigger a startScan works as expected and BleScanResults are returned.

rohitsangwan01, 2024-03-19:
@BikeBr0 Like mentioned in your logs
`Unhandled Exception: PlatformException(Bluetooth is not available, , null, null)`
Make sure to start scan only after Bluetooth is initialized, for this you can use [bluetooth-availability](https://github.com/Navideck/universal_ble?tab=readme-ov-file#bluetooth-availability) apis

BikeBr0, 2024-03-19:
Ahhh! Thank you very much @rohitsangwan01, I missed that detail (that is clearly on the readme...). So far so good when using that API. I'll continue to test this a little more before closing the ticket.

BikeBr0, 2024-03-19:
Seems to be working well for the most part. I do sometimes notice (inconsistently) that my flutter app will consider bluetooth to be off when it is in fact on. This seem so to happen most often when I run flutter in `--release` mode and shut the app down through the window's close button or user ctrl+c from the terminal (instead of q). I'll open a new ticket if I figure out the reproduction steps.

closed 2024-03-19 by BikeBr0