Navideck/universal_ble #31: Linux: Device not reporting all services

mortenboye, 2024-03-24:
I am trying to use Universal_Ble on Linux and have observed the following issue:

When scanning for devices with filtering on services:
`UniversalBle.startScan(scanFilter: ScanFilter(withServices: [MY_SERVICE_ID]));`

The intended devices are being discovered and I can connect to them. However, when discovering their services, the service which was used for the filtering of the scan is not being reported.
In my case only two services are listed:
```
flutter: Service: 0000180a-0000-1000-8000-00805f9b34fb
flutter: Service: 00001801-0000-1000-8000-00805f9b34fb
```
And I expected to see a third with the uuid corresponding to my `MY_SERVICE_ID` value.

mortenboye, 2024-03-25:
After further troubleshooting and googling, this seems to be an issue with my BlueZ installation.
Will close this issue for now.

mortenboye, 2024-03-25:
Reopened, as it appears that BlueZ does in fact see the services correctly.

This from the terminal after connecting to my device:
```
[CHG] Device CB:81:F4:3B:FB:4F UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device CB:81:F4:3B:FB:4F UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device CB:81:F4:3B:FB:4F UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device CB:81:F4:3B:FB:4F UUIDs: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[CHG] Device CB:81:F4:3B:FB:4F ServicesResolved: yes
```

How can it be that I don't see the same through Universal_ble?

mortenboye, 2024-03-26:
Reclosing. This is a Linux configuration issue.

closed 2024-03-26 by mortenboye

fotiDim, 2024-03-26:
@mortenboye what kind of configuration was needed? It might be useful for others.

mortenboye, 2024-03-26:
> @mortenboye what kind of configuration was needed? It might be useful for others.

I'm still not completely sure. Trying to work with BLE MIDI devices and not interested in using the builtin BlueZ profiles.

fotiDim, 2024-03-26:
That's a cool use case. Unfortunately I don't have any BLE MIDI device to check. From what I read the latest bluez versions support MIDI. If you manage to find what kind of configuration is needed let's document it.