Navideck/universal_ble #33: Linux: only able to receive data from one BLE device at the time.

mortenboye, 2024-04-02:
On Linux using BlueZ 5.73 I can successfully discover and connect to multiple BLE devices.

However when I connect to the second one, I stop receiving data from the first. 

I have set notifiable to `notification` and data comes in until the second device connects and starts notify as well, then `UniversalBle.onValueChanged` stops being called for the first device.

Monitoring with `bluetoothctl` and `btmon` I can verify that both devices are still connected and data are being received from both of them. However through universal_ble I only receive data from the second device.

I haven't tried with more than two devices at this point.

fotiDim, 2024-04-08:
@mortenboye thank you for the report. Have you checked if this works on any of the other platforms?

mortenboye, 2024-04-09:
> @mortenboye thank you for the report. Have you checked if this works on any of the other platforms?

I haven't tested all platforms, but it does work on Windows.

fotiDim, 2024-04-09:
OK, good to know. Probably, it should be an easy fix. We will revisit Linux in a few days time, so please hold on. In the meantime PRs are welcome!

mortenboye, 2024-04-09:
I believe I fixed it in https://github.com/Navideck/universal_ble/pull/35

fotiDim, 2024-04-10:
@mortenboye thanks for the PR. `0.9.7` was released which includes the fix. Please confirm that it works for you.

mortenboye, 2024-04-14:
It works. Great!

closed 2024-04-14 by mortenboye