Navideck/universal_ble #34: How did you make example working on iOS without modify post_install section

Wicz3k, 2024-04-08:
Hi I tried to add your lib to my project, but on iOS It was impossible without modify of post_install section of Podfile like this:
```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    flutter_additional_ios_build_settings(target)

    target.build_configurations.each do |config|
      config.build_settings['GCC_PREPROCESSOR_DEFINITIONS'] ||= [
        '$(inherited)',

        ## dart: PermissionGroup.bluetooth
        'PERMISSION_BLUETOOTH=1',
      ]

    end
  end
end
```

How your example works without this?

fotiDim, 2024-04-08:
@Wicz3k have you added `Privacy - Bluetooth Always Usage Description` in your info.plist?

Wicz3k, 2024-04-09:
@fotiDim Yes, I set this and also Bluetooth peripheral usage description. Only when I read about Permission handler I found information about modify post install section

fotiDim, 2024-04-09:
What you refer to is permission_handler specific configuration. The bluetooth permission will be requested without permission_handler as long as you have the info.plist entry, which I mentioned.

closed 2024-05-28 by fotiDim