Navideck/universal_ble #36: Only receive notifications from the first characteristic registered (Android only)

leemind, 2024-04-15:
I have an ESP32 device which notifies on 2 different characteristics.  I register both characteristics to notify

`UniversalBle.setNotifiable(devices[deviceID]!.device!.deviceId, Uuids.mainService, Uuids.firmwareControlCharStr, BleInputProperty.notification);
    UniversalBle.setNotifiable(devices[deviceID]!.device!.deviceId, Uuids.mainService, Uuids.dataCharStr, BleInputProperty.notification);` and in the debug Gatt messages this appears to be ok:

`D/BluetoothGatt(24436): setCharacteristicNotification() - uuid: f645686e-cee8-418b-971a-366c641544bc enable: true
D/BluetoothGatt(24436): setCharacteristicNotification() - uuid: 66ca5b73-7c39-4a6b-8915-ca0621b31803 enable: true`

However in my `UniversalBle.onValueChanged` I only ever receive notifications for the first registered characteristic.  I have swapped the two lines around in the code and verified this.

On iOS it works perfectly as expected

Flutter Doctor:
`[✓] Flutter (Channel stable, 3.19.5, on macOS 14.4.1 23E224 darwin-arm64, locale en-GB)
[✓] Android toolchain - develop for Android devices (Android SDK version 34.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 15.2)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2023.2)
[✓] VS Code (version 1.88.1)
[✓] Connected device (4 available)
[✓] Network resources`

rohitsangwan01, 2024-04-30:
@leemind i tried multiple android characteristic together in android, and getting events from all properly
can you try with another app like NrfConnect to cross check

sonvu1969, 2024-09-16:
Ok

rohitsangwan01, 2024-12-05:
@leemind I am still unable to reproduce this issue. I've tested various cases across different devices.

The only scenario that caused a failure for me was when attempting to subscribe to characteristics without using await or implementing a queue.

To resolve this, you need to either:

- Use `await` before subscribing to another characteristic, or
- Set up a [queue](https://github.com/Navideck/universal_ble?tab=readme-ov-file#command-queue), either globally or per device.

Without this, Android cannot handle concurrent operations, causing subsequent commands to fail. This might be the issue you're experiencing.

fotiDim, 2026-10-15:
Closing due to lack of feedback. We can reopen if the issue is still there.

closed 2026-10-15 by fotiDim