Navideck/universal_ble #47: Any way to allow update RSSI of detected devices on IOS

Wicz3k, 2024-06-04:
The default behavior of all platform is to call onScanResult after every detected advertisement packet. On ios in default callback is called only one time.
The reason is that there is no `CBCentralManagerScanOptionAllowDuplicatesKey` flag in `manager.scanForPeripherals()`.
Do you plan any options of set this flag?
Actually only way is to call start scanning on ios in loop as often as I want to have updates on RSSI of detected devices.
I know that this flag does not working in background ble scanning, so even if it will be enabled, nothing will change in background scanning.

fotiDim, 2024-06-11:
@Wicz3k it will be included in the next release, bound to happen until end of week. I confirmed the behavior on Android. Did you try on Windows or Linux?

Wicz3k, 2024-06-11:
I check Windows, MacOs, Android and iOS. Only iOS need scanning loop to update rssi.
I wasn't try Linux, maybe in future but on virtual machine when i find my USB Bletooth adapter.

I will also check if this will not change stability of ble connection. I realise that my flutter app got near totally stable connection in compare to native swift app I have previously develop.

closed 2024-06-12 by fotiDim

fotiDim, 2024-06-21:
Released in [0.10.0](https://pub.dev/packages/universal_ble).