Navideck/universal_ble #56: onConnectionChanged not called on linux

tadejg, 2024-06-27:
I'm relying on the `onConnectionChanged()` callback to detect when a device has successfully been connected. This works well on Android and web, but on Linux, the callback doesn't appear to be invoked. Call to `UniversalBle.connect()` takes a few seconds to complete, then nothing. I would expect my callback to be invoked at some point after the `UniversalBle.connect()` future completes.

rohitsangwan01, 2024-06-28:
@tadejg if your device is connecting fine, and you are not getting connection events, then seems like Connection updates are not captured for you somehow, We are using Bluez for Linux, can you give this simple bluez example a try

Create a simple dart project with : ` dart create test_bluez`
Add bluez in dependency : https://pub.dev/packages/bluez
And run this code

```dart
import 'package:bluez/bluez.dart';

void main() async {
  var client = BlueZClient();
  await client.connect();
  await _waitForAdapter(client);

  BlueZAdapter adapter = client.adapters.first;

  client.deviceAdded.listen((BlueZDevice device) {
    // Get your device here
    if (device.name == "Your Device Name") {
      adapter.stopDiscovery();
      _deviceFound(device);
    }
  });

  adapter.startDiscovery();
}

Future<void> _deviceFound(BlueZDevice device) async {
  print("DeviceFound");

  // Track connection changes
  device.propertiesChanged.listen((properties) {
    for (final property in properties) {
      if (property == "Connected") {
        print("Connection Update: ${device.connected}");
      } else {
        print("Property Update: $property");
      }
    }
  });

  // Try connection
  await device.connect();
}

Future<void> _waitForAdapter(BlueZClient client) async {
  if (client.adapters.isNotEmpty) return;

  int attempts = 0;
  while (attempts < 10 && client.adapters.isEmpty) {
    await Future.delayed(const Duration(milliseconds: 100));
    attempts++;
  }

  if (client.adapters.isEmpty) {
    throw Exception('Bluetooth adapter unavailable');
  }
}

```

tadejg, 2024-06-28:
@rohitsangwan01 Looks like that is the case. The above code gives me the same behavior - hangs at "DeviceFound". Thank you for pointing me in the right direction. I think this issue can be closed, since it's not a problem with universal_ble.

closed 2024-06-28 by tadejg