Navideck/universal_ble #69: Windows discoverServices AccessDenied

Erhannis, 2024-07-30:
I'm having a problem like #64 , but in my case the example app does display the same problem, and the BluetoothLeExplorer does not.  (Initially, anyway - but see after screenshots.)  Windows 11.  Trying to connect to a Polar Verity.  I start the example app, go to Connected Devices, select my device, click Connect, and I only see 4 services, out of 9 reported by BluetoothLeExplorer.  Here's the logs:
```
Launching lib\main.dart on Windows in debug mode...
Building Windows application...
√  Built build\windows\x64\runner\Debug\universal_ble_example.exe.
Debug service listening on ws://127.0.0.1:57690/M761VLcxEsw=/ws
Syncing files to device Windows...
[UniversalBle] Queue global
ConnectionLog: Device found
ConnectionLog: Services discovered
Failed to get characteristics for service: 0000feee-0000-1000-8000-00805f9b34fb, With Status: AccessDenied
Failed to get characteristics for service: 0000180d-0000-1000-8000-00805f9b34fb, With Status: AccessDenied
Failed to get characteristics for service: fb005c20-02e7-f387-1cad-8acd2d8df0c8, With Status: AccessDenied
Failed to get characteristics for service: 6217ff4b-fb31-1140-ad5a-a45545d7ecf3, With Status: AccessDenied
Failed to get characteristics for service: fb005c80-02e7-f387-1cad-8acd2d8df0c8, With Status: AccessDenied
ConnectionLog: Connected
flutter: _handleConnectionChange a0:9e:1a:a9:3e:4a, true
flutter: 4 services discovered
```

Here's screenshots of the list of services discovered in BluetoothLeExplorer:

[image]
[image]

...Though, I wondered why it was in "Connected Devices", and remembered that I had it open in the example app on the web page, while I tested both the desktop example app and BluetoothLeExplorer.  After closing the web page, I tried again, and this time it worked.  Logs:

```
Launching lib\main.dart on Windows in debug mode...
Building Windows application...
√  Built build\windows\x64\runner\Debug\universal_ble_example.exe.
Debug service listening on ws://127.0.0.1:61139/eOemED2qW_c=/ws
Syncing files to device Windows...
[UniversalBle] Queue global
[log] BleDevice: deviceId: 80:7a:bf:e2:01:51, name: HTC BS 519A7B, rssi: -84, isPaired: null, services: [0000cb00-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:01:51, name: HTC BS 519A7B, rssi: -83, isPaired: null, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: null, rssi: -65, isPaired: null, services: [0000180d-0000-1000-8000-00805f9b34fb, 0000feee-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -65, isPaired: null, services: [], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -81, isPaired: null, services: [0000cb00-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -80, isPaired: null, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -67, isPaired: false, services: [], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -67, isPaired: false, services: [0000180d-0000-1000-8000-00805f9b34fb, 0000feee-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -67, isPaired: false, services: [], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -78, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -77, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -78, isPaired: true, services: [0000cb00-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -77, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -81, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -81, isPaired: true, services: [0000cb00-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -81, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -65, isPaired: false, services: [], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -65, isPaired: false, services: [0000180d-0000-1000-8000-00805f9b34fb, 0000feee-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -65, isPaired: false, services: [], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -81, isPaired: true, services: [0000cb00-0000-1000-8000-00805f9b34fb], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -81, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[log] BleDevice: deviceId: d8:13:ca:57:bf:c1, name: null, rssi: -80, isPaired: null, services: [], isSystemDevice: null, manufacturerDataHead: [76, 0, 18, 25, 16, 116, 121, 207, 82, 16, 236, 154, 73, 42, 241, 126, 167, 187, 14, 88, 38, 60, 227, 61, 160, 222, 110, 2, 114], manufacturerData: [76, 0, 18, 25, 16, 116, 121, 207, 82, 16, 236, 154, 73, 42, 241, 126, 167, 187, 14, 88, 38, 60, 227, 61, 160, 222, 110, 2, 114]
[log] BleDevice: deviceId: d8:13:ca:57:bf:c1, name: null, rssi: -81, isPaired: null, services: [], isSystemDevice: null, manufacturerDataHead: [76, 0, 18, 25, 16, 116, 121, 207, 82, 16, 236, 154, 73, 42, 241, 126, 167, 187, 14, 88, 38, 60, 227, 61, 160, 222, 110, 2, 114], manufacturerData: [76, 0, 18, 25, 16, 116, 121, 207, 82, 16, 236, 154, 73, 42, 241, 126, 167, 187, 14, 88, 38, 60, 227, 61, 160, 222, 110, 2, 114]
[log] BleDevice: deviceId: a0:9e:1a:a9:3e:4a, name: Polar Sense A93E4A2A, rssi: -67, isPaired: false, services: [], isSystemDevice: null, manufacturerDataHead: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86], manufacturerData: [107, 0, 114, 8, 0, 0, 0, 0, 0, 0, 0, 0, 122, 1, 5, 51, 0, 86]
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -78, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
[... cut ...]
[log] BleDevice: deviceId: 80:7a:bf:e2:02:f4, name: HTC BS F47E45, rssi: -77, isPaired: true, services: [], isSystemDevice: null, manufacturerDataHead: [], manufacturerData: []
ConnectionLog: Device found
ConnectionLog: Services discovered
ConnectionLog: Connected
flutter: _handleConnectionChange a0:9e:1a:a9:3e:4a, true
flutter: 9 services discovered
Application finished.
```

Still seems like a strange failure mode, to report only certain services if you have more than one copy of the app open, particularly when BluetoothLeExplorer still reports all of them.

Also, my own app still reports AccessDenied - do you know what the key difference is?

rohitsangwan01, 2024-07-30:
The Windows WinRT BLE APIs can be challenging to use in a non-UWP app because you can only retrieve characteristics or services once from the API on a Bluetooth device object unless you dispose of it. This means that if you call `getCharacteristics` more than once on the same BluetoothDevice object, you will encounter issues.

In Windows, discovering services is necessary to establish a connection. Therefore, we discover services and their characteristics when we want to connect and cache them for later use. This [StackOverflow answer](https://stackoverflow.com/a/71629360) provides a clear explanation of this process.

The BluetoothLeExplorer is a UWP app, which is likely why it can discover all the services without any issues.

Erhannis, 2024-08-01:
Well, I had a call to `connect` occurring right after another, and making the second wait before retrying on failure/timeout seems to have fixed that problem.  Still seems like strange behavior.  Resolutions that seem feasible to me: 1. Discard or queue up or merge simultaneous connection attempts.  2. Don't automatically discover services on connect.  Or 3. Merge (automatic?) service discoveries, such that only the first is performed and all others use the result (plus or minus reattempts on failure, perhaps).

rohitsangwan01, 2024-08-01:
@Erhannis thanks for the suggestions, we will reconsider connection approach on windows to prevent these issues
For now, you can always call disconnect method before connect

Erhannis, 2024-09-05:
Ok, thanks.  After a lot of further work, btw - I don't know what the solution to this is, exactly, but the process is VERY finicky.  In general:
1. You MUST scan for a device before connecting to it.
2. You MUST scan for services after connecting to a device, before notifying on them, after ANY disconnect and reconnect (intentional disconnect, peripheral reboot, connection drop, or presumably other) even if you scanned them on a previous connection.
3. On Android, _sometimes_ you MUST wait between writing two things to the same characteristic (inferred from previous experience and also the fact that I was getting weird problems on Android until I increased several existing delays from 1 second to 2 seconds.)

Part of the problem is that these things appear to _vary between platforms_ - so your code will work fine on e.g. Windows but mysteriously fail on Android, and fail _differently_ on Mac.  The package may or may not tell you you've done one of these things wrong, as well - and given that a certain workflow may be valid on one platform, it's open to debate whether the one-size-fits-all approach should be _enforced_.  Given the dozens of hours I've spent debugging it, though, I'd lean strongly towards it being made the _default_, at least.

rohitsangwan01, 2024-09-06:
@Erhannis i agree there are some behaviour differences on platforms right now, and solution is to follow the same pattern on all platforms, like you mentioned, so for now flow must be

Scan for device
Connect 
Discover Services

Then we are ready to perform all operations
on Disconnect, its better to repeat above steps

For issues like you facing on Android, we have queue, maybe try to play with queue
You can open a new discussion instead, we can discuss Plugin Structure there

Our main goal was to make it simpler, that you can call any api with just the deviceId, but that came with some limitation like above, which leads to some confusion,
Another approach will be, the discovered BleDevice will have all the methods like connect and all
and similarly you have to discover services, and Services objects will have all the operations

so that you will be forced to first get the BleDevice object using Scan, and then Services object using discoverServices
But then you lose all the freedom of using Api's independently, That's subjective, we can disuss this further in [discussions](https://github.com/Navideck/universal_ble/discussions)

Erhannis, 2024-09-10:
Re: queue, the README indicates that the queue is enabled by default, but also mentions "Keep in mind that some platforms (e.g. Android) may not handle well devices that fail to process consecutive commands without a minimum interval. Therefore, it is not advised to set queueType to none."  This suggests that the default queue creates a (sufficient) minimum interval between calls.  However, I had the default queue (presumably) enabled, yet needed an additional 2-second delay between calls on Android in order for them to work.  (I'd previously had a 1 second delay which stopped being sufficient when I transitioned to universal_ble.)  Any idea why?

rohitsangwan01, 2024-09-10:
It’s difficult to determine whether the issue stems from your peripheral limit or the platform itself. The default queue should account for platform limits, but it may help to cross-check behaviour on other platforms to see if this is specific to Android only

fotiDim, 2024-09-10:
> Re: queue, the README indicates that the queue is enabled by default, but also mentions "Keep in mind that some platforms (e.g. Android) may not handle well devices that fail to process consecutive [...]

@Erhannis There are 2 potential reasons why you might need a delay between commands:
- Your (Android) device cannot send commands fast enough
- Your peripheral cannot process commands fast enough and replies with an error

The queue is indeed enabled by default.  To the best of our knowledge, the queue will prevent the first case.

If your peripheral needs additional delay then this requires to be done by you. If you don't add a delay are you getting the same behavior on other Android devices and other platforms?

Erhannis, 2024-09-10:
The problem only showed up on Android - Windows, Mac, and iOS all work correctly without the delay.  (I'm having separate issues with Linux, but I think even there I saw the relevant code work.)  I can't guarantee it's not _related_ to the device itself, but it seems like it must at least be compounded by something specific to Android.  (Testing on a Pixel 4a, Android 13, btw.)

fotiDim, 2024-09-10:
@Erhannis there is a multitude of Android devices with different Bluetooth stacks so we cannot rule  out exceptions. At least on the Android devices we tried, we didn't face issues when having the queue enabled. Do you have access to another Android device in order to cross-check its behavior without delays. A recent device from Samsung would be a good comparison.
If we confirm that it is dependent on the Android device, we could introduce an interval parameter in the queue.

fotiDim, 2026-10-15:
Closing due to lack of feedback and many changes we've done since. We can reopen if the issue is still there.

closed 2026-10-15 by fotiDim