Navideck/universal_ble #70: BLE Scanning Issue on iOS

alessandrodestefani, 2024-08-02:
I have trouble making the scanning work on IOS, i can't get any event on the scanning callback. The same code on Android and Windows is working perfectly. I have added the permission in the info.plist. 

The code:

UniversalBle.onScanResult = (event) {
      debugPrint("item found!");
};
await UniversalBle.startScan();

There is something that i am missing?

rohitsangwan01, 2024-08-02:
@alessandrodestefani have you tried example app ?

alessandrodestefani, 2024-08-05:
Ok, i have tried an empty project and also the example app.
I have found out that on IOS if you don't set   UniversalBle.onAvailabilityChange = (state) {};    the first commend startScan will not fire (the global queue will register it and will empty), the second time it will work.

If you have set onAvailabilityChange it will work fine.

rohitsangwan01, 2024-12-04:
@alessandrodestefani please try with [this](https://github.com/Navideck/universal_ble/pull/103) PR

fotiDim, 2024-12-09:
Released in [0.14.0](https://pub.dev/packages/universal_ble/versions/0.14.0)

closed 2024-12-09 by fotiDim