Navideck/universal_ble #71: Unhandled Exception: PlatformException(Failed, Failed to discover services, null, null)

konsultaner, 2024-08-06:
I get an exception when trying to register a characteristik on android (11 and 14) using the latest flutter version. It does work when compiling it to linux.

```
Unhandled Exception: PlatformException(Failed, Failed to discover services, null, null)
```

This is my demo app code:

```dart
void initState() {
    super.initState();
    _startScan();
  }

  Future<void> _startScan() async {
    AvailabilityState state = await UniversalBle.getBluetoothAvailabilityState();
    if (state != AvailabilityState.poweredOn) {
      return;
    }

    await UniversalBle.stopScan();

    UniversalBle.onScanResult = (BleDevice device) async {
      if (device.name == 'AdaESP32') {
        await UniversalBle.stopScan();
///////////////
/// I debounced it because this would be called up to 4 times in parallel. I thought this might fix this error but it didn't
/////////////////////
        _debounceConnectToDevice(device);
      }
    };

    await UniversalBle.startScan();
  }

  void _debounceConnectToDevice(BleDevice device, {Duration debounceDuration = const Duration(milliseconds: 300)}) {
    // Cancel the existing timer if it exists
    if (_debounceTimer?.isActive ?? false) {
      _debounceTimer!.cancel();
    }

    // Set up a new timer to call the connect function after the debounce duration
    _debounceTimer = Timer(debounceDuration, () => _connectToDevice(device));
  }

  Future<void> _connectToDevice(BleDevice device) async {
    _device = device;
    BleConnectionState connectionState = await _device.connectionState;
    if (connectionState == BleConnectionState.disconnected) {
      await UniversalBle.connect(_device.deviceId);
      await _discoverServices(_device.deviceId);
    } else if (connectionState == BleConnectionState.connected) {
      await _discoverServices(_device.deviceId);
    }
  }

  Future<void> _discoverServices(String deviceId) async {
    List<BleService> services = await UniversalBle.discoverServices(deviceId);
    UniversalBle.onValueChange = (String deviceId, String characteristicId, Uint8List value) {
      if (characteristicId == _characteristicUuid) {
        Uint8List newJsonBuffer = Uint8List(_jsonBuffer.length + value.length);
        newJsonBuffer.setAll(0, _jsonBuffer);
        newJsonBuffer.setAll(_jsonBuffer.length,value);
        if (newJsonBuffer.isNotEmpty && newJsonBuffer.last == 0x00) {
          _jsonBuffer = Uint8List(0);
          _handleData(newJsonBuffer);
        } else {
          _jsonBuffer = newJsonBuffer;
        }
      }
[... 83 lines cut ...]
E/flutter (23653): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:949:15)
E/flutter (23653): <asynchronous suspension>
E/flutter (23653): #3      _QueuedFuture.execute (package:universal_ble/src/queue.dart:65:18)
E/flutter (23653): <asynchronous suspension>
E/flutter (23653): 
D/BluetoothGatt(23653): onClientConnectionState() - status=0 clientIf=6 device=XX:XX:XX:XX:A3:0A
D/UniversalBlePlugin(23653): onConnectionStateChange-> Status: 0, NewState: 2
D/BluetoothGatt(23653): onConnectionUpdated() - Device=XX:XX:XX:XX:A3:0A interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(23653): onConnectionUpdated() - Device=XX:XX:XX:XX:A3:0A interval=36 latency=0 timeout=500 status=0
I/ViewRootImpl@ec5313e[MainActivity](23653): onDisplayChanged oldDisplayState=2 newDisplayState=2
I/ViewRootImpl@ec5313e[MainActivity](23653): onDisplayChanged oldDisplayState=2 newDisplayState=2
I/ViewRootImpl@ec5313e[MainActivity](23653): handleAppVisibility mAppVisible = true visible = false
I/ViewRootImpl@ec5313e[MainActivity](23653): stopped(true) old = false
D/ViewRootImpl@ec5313e[MainActivity](23653): WindowStopped on com.example.app/com.example.app.MainActivity set to true
D/OpenGLRenderer(23653): CacheManager::trimMemory(20)
```

This is my flutter version output:

```shell
Flutter 3.22.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision b0850beeb2 (vor 3 Wochen) • 2024-07-16 21:43:41 -0700
Engine • revision 235db911ba
Tools • Dart 3.4.4 • DevTools 2.34.3
```

rohitsangwan01, 2024-08-07:
@konsultaner you have to wait for connection first

```dart
await UniversalBle.connect(_device.deviceId);
await _discoverServices(_device.deviceId);
```

here when call connect method, you have to wait for connection update callback in `onConnectionChange`

```dart
UniversalBle.onConnectionChange = (String deviceId, bool isConnected) {
  debugPrint('OnConnectionChange $deviceId, $isConnected');
  // Here you can discover services if connected
}
```

rohitsangwan01, 2024-08-07:
@konsultaner is example app working fine for you ?

konsultaner, 2024-08-07:
> @konsultaner is example app working fine for you?

No, I have to [activly request for permission](https://github.com/Navideck/universal_ble/blob/5ce6c3d3627fc59fa31bafb35f61ad916a09627e/example/lib/data/permission_handler.dart#L70) to see a device on android 11.

> [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(SecurityException, java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid = 10574, packageName = com.example.app, attributionTag = null, token = android.os.BinderProxy@d6be1ce, next = null }: GattService registerScanner, Cause: null, Stacktrace: java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid = 10574, packageName = com.example.app, attributionTag = null, token = android.os.BinderProxy@d6be1ce, next = null }: GattService registerScanner

I get the same error on android 14 with your example app. Seems like the manifest permission is not working for bluetooth (anymore?). 

One has to use a third party package to do permission work. Maybe you should add this to your docs or `README.MD`. Or did I overread :exploding_head: ?

Thanks for your advice! :pray:

konsultaner, 2024-08-07:
> @konsultaner is example app working fine for you ?

The example app shows the same error.

[image]

konsultaner, 2024-08-07:
I have an idea. could it be the case that sending a '\0' for message termination might cause the issue? I'm sending a zero terminated string in 20byte chunks. In that screenshot behind the read I see the last chunk:

```shell
"tes0903co2":-1}\0
```

rohitsangwan01, 2024-08-07:
@konsultaner yes we have to ask bluetooth permissions using another plugin, you can use [this](https://github.com/Navideck/universal_ble/blob/main/example/lib/data/permission_handler.dart) class from our example app for permission part

rohitsangwan01, 2024-08-07:
> The example app shows the same error.

can you try in [NRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en_IN) app, if you are able to subscribe to this char ?

rohitsangwan01, 2024-08-07:
ok seems like you peripheral does not have the descriptor for that characteristic to start notification,  we will add a workaround to fix this issue

konsultaner, 2024-08-07:
> ok seems like you peripheral does not have the descriptor for that characteristic to start notification, we will add a workaround to fix this issue

Can I fix this on the hardware side? I'm on an `Adafruit ESP32 Feather`

rohitsangwan01, 2024-08-07:
> Can I fix this on the hardware side? I'm on an `Adafruit ESP32 Feather`

yes you can add Client Characteristic Configuration Descriptor `00002902-0000-1000-8000-00805f9b34fb` to the characteristic which you want to subscribe

closed 2024-08-07 by konsultaner

rohitsangwan01, 2024-08-07:
@konsultaner Great, btw can you try your previous code ( Without CCCD ) with this branch https://github.com/Navideck/universal_ble/pull/72

konsultaner, 2024-08-07:
ok, I'll give it a try tomorrow :smiley: I need to release my demo first.

konsultaner, 2024-08-17:
sorry, I had no chance to test it yet. I do not have the hardware at the moment. As soon as I get it back, I will give it a try!