Navideck/universal_ble #77: Under android, you cannot connect again after three or five times of repeated connection and disconnection. status 257

liwuchao-cycmotor, 2024-08-27:
Under android, you cannot connect again after three or five times of repeated connection and disconnection. status 257.

D/BluetoothGatt(27725): connect() - device: F1:ED:FC:C3:10:A9, auto: false, eattSupport: false
D/BluetoothGatt(27725): registerApp()
D/BluetoothGatt(27725): registerApp() - UUID=6047c621-6d5b-4d96-95e6-5e0793674b62
D/BluetoothGatt(27725): onClientRegistered() - status=257 clientIf=0
D/UniversalBlePlugin(27725): onConnectionStateChange-> Status: 257, NewState: 0

rohitsangwan01, 2024-12-04:
Its android limitation, we can't do anything about it,
Feel free to reopen..

rohitsangwan01, 2024-12-09:
@liwuchao-cycmotor  Should be fixed in https://github.com/Navideck/universal_ble/pull/109/files

fotiDim, 2024-12-09:
Released in [0.14.0](https://pub.dev/packages/universal_ble/versions/0.14.0)

closed 2024-12-09 by fotiDim

liwuchao-cycmotor, 2024-12-10:
Thank you very much!