Navideck/universal_ble #79: Remove "Characteristic already notifying" error

Erhannis, 2024-09-05:
On at least some platforms, trying to notify twice on a characteristic throws an error.  IMO, it should not.  The user wants to be notified of changes to a characteristic - if that's already happening, nothing needs to change, no error needs to be thrown, and success is automatic.  Throwing an error suggests notifications are NOT happening.  Is there a counterpoint I've missed, a reason to continue throwing that error?

I see the error in Linux, and in the Web code, but if the error is present elsewhere, this issue addresses them, as well.

rohitsangwan01, 2024-09-06:
@Erhannis, I see what you mean, and I believe this is another case of platform implementation differences. We'll work on unifying the behaviour across platforms. Thank you for pointing this out!

rohitsangwan01, 2024-12-09:
@Erhannis Fixed in https://github.com/Navideck/universal_ble/pull/110

fotiDim, 2024-12-09:
Released in [0.14.0](https://pub.dev/packages/universal_ble/versions/0.14.0)

closed 2024-12-09 by fotiDim