Navideck/universal_ble #91: Windows filter with services issue

Wicz3k, 2024-10-03:
On ios and android if i set list of services to scan ScanFilter like that:
```
List<String> allScanningServices = [
        "0000fae5-0000-1000-8000-00805f9b34fb",
        "6e400001-b5a3-f393-e0a9-e50e24dcca9e",
      ];

UniversalBle.startScan(
  scanFilter: ScanFilter(
    withServices: allScanningServices 
  ),
)
```
Application accept all devices which got any of listed services. But on windows it get device only if there is one service in list, and device got this service (i do not tried when device has all services from list). I need to find devices with any of these services, but olnly way on windows is to scan without this filter. This issue is in library since I remember.

fotiDim, 2024-10-03:
@Wicz3k please try with version 0.13 which was released earlier today.

Wicz3k, 2024-10-03:
I tried also on new version and nothing changed.
Last update make me to reviev my code and I found that I do not use this filter on windows.

fotiDim, 2024-10-07:
@Wicz3k is that resolved then? Can this be closed?

Wicz3k, 2024-10-07:
@fotiDim No, nothing changed. I still need workaround to detect my devices on windows. Workaround is to not use filter on windows and filter devices after detect them.

rohitsangwan01, 2024-10-07:
@Wicz3k are you only using service filter ?
If yes, try to add one more filter like Name filter, This will help us to narrow down the issue
After applying one more filter, Left us know if its working as expected or not

Wicz3k, 2024-10-07:
@rohitsangwan01 If I add any name prefix which does not exist near me it works as expected.

rohitsangwan01, 2024-10-07:
@Wicz3k Thank you for your cooperation. We'll resolve this issue in the next update. In the meantime, you can use a workaround by adding a random name prefix filter.

rohitsangwan01, 2024-12-09:
@Wicz3k fixed in https://github.com/Navideck/universal_ble/pull/111

fotiDim, 2024-12-09:
Released in [0.14.0](https://pub.dev/packages/universal_ble/versions/0.14.0)

closed 2024-12-09 by fotiDim