Navideck/universal_ble #92: [0.13.0] "UniversalBle.connect" seems to be fully broken for iOS?

AlexDochioiu, 2024-10-19:
I tried to upgrade universal_ble for [our package](https://github.com/vespr-wallet/ledger-flutter-plus) from version 0.12.0 to 0.13.0 and it seems that `connect` just does not work anymore. Behaviour observations:

[0.12.0][iOS] Calling await for `UniversalBle.connect` causes it to hang and not complete.
[0.12.0][iOS] Calling `UniversalBle.connect` without awaiting seems to work perfectly fine (When connection completes, I get notified via the listener added to `UniversalBle.onConnectionChange`)

[0.13.0][iOS] Calling await for `UniversalBle.connect` causes it to hang and not complete.
[0.13.0][iOS] Calling `UniversalBle.connect` without awaiting seems to still not work, unlike 0.12.0. `UniversalBle.onConnectionChange` never gets invoked.

rohitsangwan01, 2024-10-25:
@AlexDochioiu now `connect` will either complete or throw error if failed to connect, can you please cross check

AlexDochioiu, 2024-10-28:
@rohitsangwan01 On version 0.13.0? It just seems to always fail to connect and time out with error along the lines of:

PlatformException(IllegalArgument, Unknown deviceId: DE:F1:2C:9F:41:60, null, null)

fotiDim, 2024-10-28:
@AlexDochioiu can you try the example app and verify that it works? Can you also cross-check with another type of peripheral?

AlexDochioiu, 2024-10-28:
I will do that. But, for context. The exact same device (and device type) works consistently on 0.12.0 and never works on 0.13.0.

fotiDim, 2024-10-28:
0.13 was a big release. We did plenty of changes but we are also actively using it with multiple hardware configurations and didn't face any issue. Let's try to narrow it down and see where the problem lies.

rohitsangwan01, 2024-10-31:
@AlexDochioiu in your code, now you don't have to handle all the complexity of waiting for the connection from callbacks [here](https://github.com/vespr-wallet/ledger-flutter-plus/blob/master/lib/src/ledger/ledger_ble_manager.dart#L115)

Now all this is handled within UniversalBle and you can just call `connect` api and await for the completion

```dart
try {
  print("Connecting");
  await UniversalBle.connect(
    deviceId,
    connectionTimeout: const Duration(seconds: 5),
  );
  print("Connected successfully");
} catch (e) {
  print("Connection Failed with error: $e");
}
```

AlexDochioiu, 2024-11-01:
Sorry for being slow to reply. I tried that and it did not work for me. I believe the await to connect was always failing with platform error: unknown device id for some reason. I'll need to update again to 0.13 to double check since it's been a while and my memory is a bit foggy.

rohitsangwan01, 2024-11-02:
@AlexDochioiu To avoid this error `unknown device id` , you need to perform a scan before attempting a connection. Only after scanning can you interact with the device. This applies to previous versions as well.

AlexDochioiu, 2024-11-02:
@rohitsangwan01 I am performing a scan. That's how I get the device id in the first place. It's not hard coded/etc.

rohitsangwan01, 2024-12-04:
@AlexDochioiu, could you try the new version on Main and let me know the current status? As I mentioned earlier, we can now simply wait for the connection attempt to either complete or fail with an error. The [connect](https://github.com/vespr-wallet/ledger-flutter-plus/blob/master/lib/src/ledger/ledger_ble_manager.dart#L58C1-L125C1) logic can now be streamlined. Please let me know if you still encounter any issues.

Here’s how you can modify your connect method:

```dart
@override
Future<void> connect(LedgerDevice device) async {
    ....
    
    // No need for completers, just call connect and await the result
    try {
        print("Connecting");
        await UniversalBle.connect(
           deviceId,
          connectionTimeout: const Duration(seconds: 5),
       );
       // Logic after successful connection...
    } catch (e) {
      // Handle failure...
    }
    ....
}
```

fotiDim, 2024-12-09:
@AlexDochioiu could you please rest the newly released 0.14 with the above code suggestion and report back?

AlexDochioiu, 2025-01-14:
Hey @fotiDim apologies for the slow reply. It's been a really busy period. I'll do my best to try 0.14 asap and report back.

AlexDochioiu, 2025-01-15:
@fotiDim Still no luck. With version 0.14 the "UniversalBle.connect" hangs while I am awaiting for it. Additionally, I can see the below in XCode logs.

[image]

rohitsangwan01, 2025-01-15:
@AlexDochioiu Could you clarify what you mean by "hang"?
Are you receiving the `.disconnected` event, or does the connect API fail to respond in cases of disconnection or connection failure? If the latter, it should ideally timeout.

Have you been able to reproduce the same behavior in the example app?
Or If possible, could you provide minimal reproducible code to help us investigate further?

rohitsangwan01, 2025-01-15:
@AlexDochioiu if any error occurs in connection, then it should be captured on dart side in catch block, can you try to wrap connect method with try/catch block and see if you are getting any error

AlexDochioiu, 2025-01-15:
It is already wrapped in a try/catch @rohitsangwan01 . The `connect` method throws just the timeout exception, despite of the error showing up in the XCode.

rohitsangwan01, 2025-01-15:
@AlexDochioiu Seems like some kind of race condition in calling connect and capturing the error, are you also listening for the connection update callback ?
if yes, that callback also returns the error message if available when disconnect, can you check if you are getting any error message in it

AlexDochioiu, 2025-01-15:
@rohitsangwan01 Interestingly, it seems that I can establish connection using the test app. I removed the callback for `UniversalBle.onConnectionChange` when I migrated from 0.12 to 0.14.

AlexDochioiu, 2025-01-15:
Interesting turn of events. After adding the `onConnectionChange` listener, connection seems to work again.

[image]

rohitsangwan01, 2025-01-15:
> Interesting turn of events. After adding the onConnectionChange listener, connection seems to work again.

Thats interesting, we have to check on this behaviour

AlexDochioiu, 2025-01-15:
Found the issue @rohitsangwan01 . I'll raise a PR to fix it. It's definitely an odd one.

fotiDim, 2025-01-15:
@AlexDochioiu we merged a [PR](https://github.com/Navideck/universal_ble/pull/130) which might resolve your issue. Could you please try again without any code changes on your side and see if you still have connection trouble? Just put this in your  pubspec:
```yaml
universal_ble:
    git:
      url: https://github.com/Navideck/universal_ble.git
      ref: main
```

AlexDochioiu, 2025-01-16:
@fotiDim It will work for sure looking at the changes in the merged PR. Thank you!

I will go ahead and close my PR since the changes you merged are covering what was in my PR.

closed 2025-01-16 by AlexDochioiu