Navideck/universal_ble #94: how to send messages using universal_ble between two mobiles ?

coybitcom, 2024-11-03:
how to send messages using universal_ble between two mobiles ?

coybitcom, 2024-11-03:
The peer party did not monitor the data when sending the message.
This configuration is invalid.

  @override
  void initState() {
    super.initState();
    UniversalBle.onConnectionChange = _handleConnectionChange;
    UniversalBle.onValueChange = _handleValueChange;
    UniversalBle.onPairingStateChange = _handlePairingStateChange;
  }

rohitsangwan01, 2024-11-03:
@coybitcom To enable communication between two devices over BLE, one device needs to act as a GATT Server and the other as a GATT Client.

For handling the Client-side BLE communication, you can use UniversalBle. On the other device, you’ll need to set up a GATT Server (peripheral). You can use libraries like [ble_peripheral](https://pub.dev/packages/ble_peripheral) to create a BLE peripheral for this purpose.

coybitcom, 2024-11-03:
Thank you very much for your valuable advice

closed 2024-11-03 by coybitcom