Navideck/universal_ble #95: How to get Service Data?

ApparentWindDirection, 2024-11-04:
Some background:
I am trying to get readings from a BLE temperature sensor (Blue PUCK T-PROBE).
The readings are transmitted through advertising packets, encoded as a sequence of bytes inside the [Service Data](https://www.bluetooth.com/blog/advertising-works-part-2/) field for the standard temperature service with 16-bit UUID `2a6e`. Apparently, this format is standardized in [Environmental Sensing Service](https://www.bluetooth.com/specifications/specs/environmental-sensing-service-1-0/). Getting readings should be completely passive and doesn't require establishing a connection to the sensor. Indeed, it can be passively read both with nRF Connect and with custom code using [win_ble](https://pub.dev/packages/win_ble).

I was unable to find a way to get readings with this library. 
I've tried discovering services of the sensor without establishing a connection, but no success. [Documentation](https://pub.dev/documentation/universal_ble/latest/index.html) is clear on that saying `You need to first discover services before you are able to read and write to characteristics.`. Yet the sensor isn't meant to be connected.
I've tried discovering services after connecting to the sensor. Discovery finishes successfully, but the list of discovered services doesn't contain the temperature service `2a6e`. Apparently, parsing Service Data is the only way to get temperature readings. In any case, even if it would be possible to get them after a connection, reading a dozen of sensors (with a practical limit on the number of simultaneous connections in mind) doesn't seem reasonable.

Is there any way to retrieve raw bytes from Service Data fields in advertisement packets? Specifically, without a connection to the target device.

rohitsangwan01, 2024-11-06:
@ApparentWindDirection ServiceData is currently not exposed in scan results, because platform support for it is limited

ApparentWindDirection, 2024-11-07:
@rohitsangwan01 It is unfortunate that this feature is not available, since it precludes not only environmental sensors from the aforementioned standard, but a  [whole class](https://en.wikipedia.org/wiki/Bluetooth_Low_Energy_beacon) of BLE devices from being used. 
Is platform support limited, though? Quick googling shows that Service Data is available on [Android](https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getServiceData()), [iOS & macOS](https://developer.apple.com/documentation/corebluetooth/cbadvertisementdataservicedatakey), [Windows](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementdatasection?view=winrt-26100), [Linux](https://pub.dev/documentation/bluez/latest/bluez/BlueZDevice/serviceData.html) and planned on [Web](https://webbluetoothcg.github.io/web-bluetooth/#bluetoothservicedatamap). It is already  better supported than `unpair` API.

rohitsangwan01, 2024-11-07:
@ApparentWindDirection Thank you for the feedback. We’ll look into adding support for this feature and aim to include it in an upcoming update.

FluffyBunniesTasteTheBest, 2025-07-07:
Exposing the advertised service data would also enable communication with BLE Mesh networks. 

BTW, thank you very much for providing such a solid package.