Navideck/universal_ble #96: Support for BLE extended advertising on Windows

SaltyLad, 2024-11-14:
Hello,

I've noticed that the Windows implementation of this plugin currently lacks support for BLE extended advertising. As a result, devices using extended advertising features, such as coded PHY, aren’t detected during scans.

This can be addressed by enabling BluetoothLEAdvertisementWatcher.AllowExtendedAdvertisements after the watcher is instantiated. I’d be happy to submit a pull request to make this change if you’re open to supporting this feature.

[More details here](https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementwatcher.allowextendedadvertisements?view=winrt-26100)

rohitsangwan01, 2024-11-15:
Hey, because there are multiple platforms involved, it would be great if we have some info about rest of the platforms as well, and then decide the Api behaviour

SaltyLad, 2024-11-19:
Hi, 

Sorry for the delay in responding. I have tested the package on the Android platform. I can confirm devices using extended advertising features are discovered when scanning, no changes are required.

I unfortunately cannot test for Mac or iOS as I do not have Apple devices available.

I would suggest this setting be enabled by default on Windows without necessarily adding an API option. Legacy advertising devices could still be seen during scanning, and this would bring the Windows version in line with the Android version.

rohitsangwan01, 2024-12-09:
@SaltyLad implemented here https://github.com/Navideck/universal_ble/pull/114

closed 2026-10-14 by fotiDim