Navideck/universal_ble #98: On Android 11 turning location on/off doesn't trigger onAvailabilityChange

Sravdar, 2024-11-20:
On Android 11 (and lower) location is required to be ON in order to perform scan. But turning it ON or OFF doesn't trigger onAvailabilityChange.

Tested on:
Redmi Note 8
Android 11 RKQ.201004.002

rohitsangwan01, 2024-11-20:
@Sravdar `onAvailabilityChange` is for notifying Bluetooth Radio State only,  for now you have to implement your own solution to get Location updates

rohitsangwan01, 2024-12-04:
This functionality is out of scope for this plugin. If you need to listen for location updates, consider using other plugins such as Geolocator. You can find detailed instructions on how to listen for updates [here](https://pub.dev/packages/geolocator#:~:text=Location%20service%20information&text=To%20listen%20for%20service%20status,receive%20location%20service%20status%20updates).

The onAvailabilityChange event in this plugin only notifies you about changes to the Bluetooth adapter's state (on/off), not location updates.

and you can combine both `onAvailabilityChange` and `Geolocator.getServiceStatusStream()` to achieve your desired behaviour

feel free to reopen if you have any doubt

closed 2024-12-04 by fotiDim