Navideck/universal_ble #99: On android scan doesn't start if requesting more than 5 scans within 30 second

Sravdar, 2024-11-20:
This is implemented protection by android for scan abuse. https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library/issues/18#issuecomment-402412139

It would be great if universal_ble throwed exception in these cases or handle it so it starts scaning when it can.

rohitsangwan01, 2024-12-04:
@Sravdar thanks for the info, seems like Android does not notify if scan failed because of `Scan too frequently`
so tried adding a custom solution from the link you mentioned, you can try pr https://github.com/Navideck/universal_ble/pull/104

Sravdar, 2024-12-04:
@rohitsangwan01 I've tested on my app and its working as expected. Did not encountered any issue. Thank you for your work.

Sravdar, 2024-12-04:
On my android 13 device it worked fine but on android 11 there are some problems. When app is launched while bluetooth is disabled and then when bluetooth is re-enabled manually while app is open nothing from universal_ble seems to be working. It works as expected when app is launched while bluetooth is enabled.

If it helps here is the logs i get:
**When app first opens**
```
Connected to the VM Service.
[Ble.onAvailabilityChange] AvailabilityState.poweredOff
D/ProfileInstaller( 9437): Installing profile for com.example.aybey_connect
E/LB      ( 9437): fail to open file: No such file or directory
```
**Attempt to stop then restart the scan**
```
D/ProfileInstaller( 9437): Installing profile for com.example.aybey_connect
E/LB      ( 9437): fail to open file: No such file or directory
V/MiuiFrameworkFactory( 9437): get AllImpl object = android.common.MiuiFrameworkFactoryImpl@ff67a4c
W/MirrorManager( 9437): this model don't Support
[BLE stopScan] Catched error
[BLE stopScan] PlatformException (PlatformException(BluetoothNotEnabled, Bluetooth not enabled, null, null))
[BLE restartScan] Catched error
[BLE restartScan] PlatformException (PlatformException(BluetoothNotEnabled, Bluetooth not enabled, null, null))
```
**Bluetooth is enabled** (Note that no onAvailabilityChange is triggered)
```
D/BluetoothAdapter( 9437): onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@19df7aa
```
**Going back page which attempts to disconnect last connected device and stops the scan**
```
[BLE disconnect] Catched error
[BLE disconnect] PlatformException (PlatformException(IllegalArgument, Unknown deviceId: , null, null))
[BLE stopScan] Catched error
[BLE stopScan] PlatformException (PlatformException(UninitializedPropertyAccessException, kotlin.UninitializedPropertyAccessException: lateinit property safeScanner has not been initialized, Cause: null, Stacktrace: kotlin.UninitializedPropertyAccessException: lateinit property safeScanner has not been initialized
               	at com.navideck.universal_ble.UniversalBlePlugin.stopScan(UniversalBlePlugin.kt:153)
               	at com.navideck.universal_ble.UniversalBlePlatformChannel$Companion.setUp$lambda$7$lambda$6(UniversalBle.g.kt:356)

```

rohitsangwan01, 2024-12-05:
_Originally posted by @Sravdar

On my android 13 device it worked fine but on android 11 there are some problems. When app is launched while bluetooth is disabled and then when bluetooth is re-enabled manually while app is open nothing from universal_ble seems to be working. It works as expected when app is launched while bluetooth is enabled.

If it helps here is the logs i get:
**When app first opens**
```
Connected to the VM Service.
[Ble.onAvailabilityChange] AvailabilityState.poweredOff
D/ProfileInstaller( 9437): Installing profile for com.example
E/LB      ( 9437): fail to open file: No such file or directory
```
**Attempt to stop then restart the scan**
```
D/ProfileInstaller( 9437): Installing profile for com.example.aybey_connect
E/LB      ( 9437): fail to open file: No such file or directory
V/MiuiFrameworkFactory( 9437): get AllImpl object = android.common.MiuiFrameworkFactoryImpl@ff67a4c
W/MirrorManager( 9437): this model don't Support
[BLE stopScan] Catched error
[BLE stopScan] PlatformException (PlatformException(BluetoothNotEnabled, Bluetooth not enabled, null, null))
[BLE restartScan] Catched error
[BLE restartScan] PlatformException (PlatformException(BluetoothNotEnabled, Bluetooth not enabled, null, null))
```
**Bluetooth is enabled** (Note that no onAvailabilityChange is triggered)
```
D/BluetoothAdapter( 9437): onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@19df7aa
```
**Going back page which attempts to disconnect last connected device and stops the scan**
```
[BLE disconnect] Catched error
[BLE disconnect] PlatformException (PlatformException(IllegalArgument, Unknown deviceId: , null, null))
[BLE stopScan] Catched error
[BLE stopScan] PlatformException (PlatformException(UninitializedPropertyAccessException, kotlin.UninitializedPropertyAccessException: lateinit property safeScanner has not been initialized, Cause: null, Stacktrace: kotlin.UninitializedPropertyAccessException: lateinit property safeScanner has not been initialized
               	at com.navideck.universal_ble.UniversalBlePlugin.stopScan(UniversalBlePlugin.kt:153)
               	at com.navideck.universal_ble.UniversalBlePlatformChannel$Companion.setUp$lambda$7$lambda$6(UniversalBle.g.kt:356)

```

rohitsangwan01, 2024-12-05:
@Sravdar am not able to reproduce this issue, can you check if you are getting same behaviour on Main branch ?

Sravdar, 2024-12-05:
@rohitsangwan01 Can confirm this problem didn't exist on main and can confirm your lastest commit fixed this issue. Thank you.

fotiDim, 2024-12-09:
Released in [0.14.0](https://pub.dev/packages/universal_ble/versions/0.14.0)

closed 2024-12-09 by fotiDim