Navideck/universal_ble #108: PR #106 broke disconnect functionality and probably more

Sravdar, 2024-12-05:
I've tested #106 on android 11 device and it had some problems.

I can scan, connect, sub and recieve as normal. When bluetooth turns off unexpectedly it correctly cleanup and no duplicaiton issue happens.

But when i try to disconnect `UniversalBle.disconnect()` regardless of if im subbed to char or not it doesn't properly disconnects. It does not trigger `onAvailabilityChange` while i get following logs

```
D/BluetoothGatt(24430): cancelOpen() - device: F4:12:FA:18:88:8A
D/BluetoothGatt(24430): close()
D/BluetoothGatt(24430): unregisterApp() - mClientIf=6
```

If i still try re-connect the device not checking `AvailabilityState` it seems to be reconnected but i can't sub to chars and get following error when i try to do so

```
E/flutter (29367): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(Failed, Failed to update subscription state, java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference, null)
E/flutter (29367): #0      UniversalBlePlatformChannel.setNotifiable (package:universal_ble/src/universal_ble_pigeon/universal_ble.g.dart:435:7)
E/flutter (29367): <asynchronous suspension>
E/flutter (29367): #1      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:963:15)
E/flutter (29367): <asynchronous suspension>
E/flutter (29367): #2      _QueuedFuture.execute (package:universal_ble/src/queue.dart:65:18)
E/flutter (29367): <asynchronous suspension>
E/flutter (29367): 
```

or get following error when trying to write to char without subscribing to any char.

```
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): Failed to handle message
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothGattCharacteristic android.bluetooth.BluetoothGattService.getCharacteristic(java.util.UUID)' on a null object reference
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.navideck.universal_ble.UniversalBleHelperKt.getCharacteristic(UniversalBleHelper.kt:182)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.navideck.universal_ble.UniversalBlePlugin.writeValue(UniversalBlePlugin.kt:406)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.navideck.universal_ble.UniversalBlePlatformChannel$Companion.setUp$lambda$21$lambda$20(UniversalBle.g.kt:498)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.navideck.universal_ble.UniversalBlePlatformChannel$Companion.$r8$lambda$dE18af8N73VCRd4knNHkyoeXVf0(Unknown Source:0)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.navideck.universal_ble.UniversalBlePlatformChannel$Companion$$ExternalSyntheticLambda7.onMessage(Unknown Source:2)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at io.flutter.plugin.common.BasicMessageChannel$IncomingMessageHandler.onMessage(BasicMessageChannel.java:261)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:292)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at android.os.Handler.handleCallback(Handler.java:938)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at android.os.Looper.loop(Looper.java:236)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at android.app.ActivityThread.main(ActivityThread.java:8037)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at java.lang.reflect.Method.invoke(Native Method)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
E/BasicMessageChannel#dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue(11705): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)
E/flutter (11705): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(channel-error, Unable to establish connection on channel: "dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.writeValue"., null, null)
E/flutter (11705): #0      UniversalBlePlatformChannel.writeValue (package:universal_ble/src/universal_ble_pigeon/universal_ble.g.dart:536:7)
E/flutter (11705): <asynchronous suspension>
E/flutter (11705): #1      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:963:15)
E/flutter (11705): <asynchronous suspension>
E/flutter (11705): #2      _QueuedFuture.execute (package:universal_ble/src/queue.dart:65:18)
E/flutter (11705): <asynchronous suspension>
E/flutter (11705): 
```

And after this error almost everything seems to be broken.

rohitsangwan01, 2024-12-05:
@Sravdar Thankyou for the information, can you confirm if its fixed for you in this pr https://github.com/Navideck/universal_ble/pull/109

Sravdar, 2024-12-05:
@rohitsangwan01 Yes it works without any problems now on both android 11 and 13. Problem on #15 is fixed as well.

closed 2024-12-09 by fotiDim