Navideck/universal_ble #117: Scanning Beacons on Web

ImanHz, 2024-12-14:
I am using `universal_ble` version 0.14.0 on Flutter 3.24.5 and Dart 3.5.4. 
Using the `startScan`  function on Linux, shows the beacons but using the same function on Web, fails to list the beacons. Is it a limitation in the WebBLE standard?

ImanHz, 2024-12-21:
As I investigated, the service uuid should be embedded in the advertisement packet. Not tested yet.

rohitsangwan01, 2024-12-27:
@ImanHz if you startScan without any scanFilter it should show all ble advertising devices, 
you can play with the web filter, checkout [this](https://github.com/Navideck/universal_ble?tab=readme-ov-file#web)

fotiDim, 2025-05-12:
Closing due to lack of feedback. We can reopen it if the issue persists.