Navideck/universal_ble #118: ErrorCallback: Unable to establish connection on channel when hot restart

aavaa-farnood, 2024-12-23:
I receive this error around 400 times upon hot restart of the application:

```
ErrorCallback: Unable to establish connection on channel: 'dev.flutter.pigeon.universal_ble.UniversalBleCallbackChannel.onScanResult'.
```

Flutter:

```
Flutter 3.27.1 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 17025dd882 (7 days ago) • 2024-12-17 03:23:09 +0900
Engine • revision cb4b5fff73
Tools • Dart 3.6.0 • DevTools 2.40.2
```

Package:

```
universal_ble: ^0.14.0
```

rohitsangwan01, 2024-12-27:
> I receive this error approximately 400 times upon a hot restart of the application.

Are you saying the error occurs on the 400th hot restart, or does it happen 400 times during a single hot restart? Could you clarify?

aavaa-farnood, 2025-01-02:
After one single hot restart, it throws 400 errors like mentioned above until the application restarts completely.

rohitsangwan01, 2025-01-08:
@aavaa-farnood can you share a minimal code snippet to reproduce this

rohitsangwan01, 2025-02-13:
@aavaa-farnood Which platform you are using ? and whats the OS version ?

aavaa-farnood, 2025-02-17:
Oh sorry I thought I included that, it's:

```
Windows 11 Pro
24H2
OS build 26100.3194
```

rohitsangwan01, 2025-02-21:
@aavaa-farnood Please try with https://github.com/Navideck/universal_ble/pull/142

fotiDim, 2025-05-12:
Closing due to lack of feedback. Should be resolved.