Navideck/universal_ble #119: Cannot run in windows platform

zjw78123sc, 2024-12-26:
LINK : warning LNK4217: ���š�??1bad_cast@std@@UEAA@XZ (public: virtual __cdecl std::bad_cast::~bad_cast(void))��(�ڡ� universal_ble.g.obj���ж���)���ɡ�universal_ble_filter_util.obj��(������"public: virtual __cdecl std::bad_any_cast::~bad_any_cast(void)" (??1bad_any_cast@std@@UEAA@XZ)����)���� [E:\Mine\ai_mouse\ai_mouse\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
universal_ble_filter_util.obj : error LNK2019: �޷��������ⲿ���� "__declspec(dllimport) public: __cdecl std::bad_cast::bad_cast(char const *)" (__imp_??0bad_cast@std@@QEAA@PEBD@Z)������ "public: __cdecl std::bad_any_cast::bad_any_cast(void)" (??0bad_any_cast@std@@QEAA@XZ) �������˸÷��� [E:\Mine\ai_mouse\ai_mouse\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
universal_ble_filter_util.obj : error LNK2001: �޷��������ⲿ���� "protected: virtual void __cdecl std::bad_cast::_Doraise(void)const " (?_Doraise@bad_cast@std@@MEBAXXZ) [E:\Mine\ai_mouse\ai_mouse\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
E:\Mine\ai_mouse\ai_mouse\build\windows\x64\plugins\universal_ble\Debug\universal_ble_plugin.dll : fatal error LNK1120: 2 ���޷��������ⲿ���� [E:\Mine\ai_mouse\ai_mouse\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
Error: Build process failed.

Sravdar, 2024-12-26:
Do you have lastest [Visual Studio (2022)](https://visualstudio.microsoft.com/downloads/?cid=learn-onpage-download-install-visual-studio-page-cta) installed with "Desktop development with C++"? I used to get build errors when using vs2019.

zjw78123sc, 2024-12-27:
> Do you have lastest [Visual Studio (2022)](https://visualstudio.microsoft.com/downloads/?cid=learn-onpage-download-install-visual-studio-page-cta) installed with "Desktop development with C++"? I used to get build errors when using vs2019.

yes,i use Visual Studio 2022,but still have the problem above.

rohitsangwan01, 2024-12-27:
@zjw78123sc are you using latest Flutter ?
please share output of `flutter doctor`

zjw78123sc, 2024-12-27:
> @zjw78123sc are you using latest Flutter ? please share output of `flutter doctor`

3.27.1

zjw78123sc, 2025-01-14:
flutter doctor

┌─────────────────────────────────────────────────────────┐
│ A new version of Flutter is available!                  │
│                                                         │
│ To update to the latest version, run "flutter upgrade". │
└─────────────────────────────────────────────────────────┘

  ╔════════════════════════════════════════════════════════════════════════════╗
  ║                 Welcome to Flutter! - https://flutter.dev                  ║
  ║                                                                            ║
  ║ The Flutter tool uses Google Analytics to anonymously report feature usage ║
  ║ statistics and basic crash reports. This data is used to help improve      ║
  ║ Flutter tools over time.                                                   ║
  ║                                                                            ║
  ║ Flutter tool analytics are not sent on the very first run. To disable      ║
  ║ reporting, type 'flutter config --no-analytics'. To display the current    ║
  ║ setting, type 'flutter config'. If you opt out of analytics, an opt-out    ║
  ║ event will be sent, and then no further information will be sent by the    ║
  ║ Flutter tool.                                                              ║
  ║                                                                            ║
  ║ By downloading the Flutter SDK, you agree to the Google Terms of Service.  ║
  ║ Note: The Google Privacy Policy describes how data is handled in this      ║
  ║ service.                                                                   ║
  ║                                                                            ║
  ║ Moreover, Flutter includes the Dart SDK, which may send usage metrics and  ║
  ║ crash reports to Google.                                                   ║
  ║                                                                            ║
  ║ Read about data we send with crash reports:                                ║
  ║ https://flutter.dev/docs/reference/crash-reporting                         ║
  ║                                                                            ║
[√] Android Studio (version 3.6)
[!] Android Studio (version 2022.1)
    X Unable to find bundled Java version.
[!] Android Studio (version 2022.3)
    X Unable to find bundled Java version.
[√] IntelliJ IDEA Ultimate Edition (version 2017.3)
[√] Connected device (3 available)
[√] HTTP Host Availability

! Doctor found issues in 3 categories.
@rohitsangwan01

zjw78123sc, 2025-01-14:
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
[!] Flutter (Channel stable, 3.27.1, on Microsoft Windows [版本 10.0.18363.1556], locale zh-CN)
    • Flutter version 3.27.1 on channel stable at W:\Android\flutter_windows_3.27.1-stable\flutter
    ! Warning: `flutter` on your path resolves to W:\Android\flutter_windows_3.3.0-stable\flutter\bin\flutter, which is not inside your current Flutter SDK checkout at W:\Android\flutter_windows_3.27.1-stable\flutter. Consider adding W:\Android\flutter_windows_3.27.1-stable\flutter\bin to the front of your path.
    ! Warning: `dart` on your path resolves to W:\Android\flutter_windows_3.3.0-stable\flutter\bin\dart, which is not inside your current Flutter SDK checkout at W:\Android\flutter_windows_3.27.1-stable\flutter. Consider adding W:\Android\flutter_windows_3.27.1-stable\flutter\bin to the front of your path.
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 17025dd882 (4 weeks ago), 2024-12-17 03:23:09 +0900
    • Engine revision cb4b5fff73
    • Dart version 3.6.0
    • DevTools version 2.40.2
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn
    • If those were intentional, you can disregard the above warnings; however it is recommended to use "git" directly to perform update checks and upgrades.

[√] Windows Version (Installed version of Windows is version 10 or higher)

[√] Android toolchain - develop for Android devices (Android SDK version 31.0.0-rc2)
    • Android SDK at W:\Android\sdk
    • Platform android-31, build-tools 31.0.0-rc2
    • ANDROID_HOME = W:\Android\sdk
    • ANDROID_SDK_ROOT = W:\Android\sdk
    • Java binary at: E:\Android\Android Studio\jbr\bin\java
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-10027231)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files (x86)\Google\Chrome\Application\chrome.exe

[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.1.1)
    • Visual Studio at W:\Microsoft Visual Studio\2022\Community
    • Visual Studio Community 2022 version 17.1.32228.430
    • Windows 10 SDK version 10.0.19041.0

[√] Android Studio (version 3.6)
    • Android Studio at W:\Android\android-studio-ide-192.6071332-windows\android-studio
    • Flutter plugin version 49.0.1
    • Dart plugin version 192.8052
    • Java version OpenJDK Runtime Environment (build 1.8.0_212-release-1586-b04)

[√] Android Studio (version 2022.1)
    • Android Studio at W:\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.15+0-b2043.56-9505619)

[√] Android Studio (version 2022.3)
    • Android Studio at E:\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 17.0.6+0-b2043.56-10027231)

[√] IntelliJ IDEA Ultimate Edition (version 2017.3)
    • IntelliJ at W:\IntelliJ IDEA 2017.3.5
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [版本 10.0.18363.1556]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 131.0.6778.265
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 130.0.2849.80

[√] Network resources
    • All expected network resources are available.

! Doctor found issues in 1 category.
Process finished with exit code 0

zjw78123sc, 2025-01-14:
@rohitsangwan01 @Sravdar

rohitsangwan01, 2025-01-14:
@zjw78123sc have you tried example app

zjw78123sc, 2025-01-14:
@rohitsangwan01 yes,also error
LINK : warning LNK4217: ���š�??1bad_cast@std@@UEAA@XZ (public: virtual __cdecl std::bad_cast::~bad_cast(void))��(�ڡ� universal_ble.g.obj���ж���)���ɡ�universal_ble_filter_util.obj��(������"public: virtual __cdecl std::bad_any_cast::~bad_any_cast(void)" (??1bad_any_cast@std@@UEAA@XZ)����)���� [E:\AMyDownload\universal_ble-main\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
universal_ble_filter_util.obj : error LNK2019: �޷��������ⲿ���� "__declspec(dllimport) public: __cdecl std::bad_cast::bad_cast(char const *)" (__imp_??0bad_cast@std@@QEAA@PEBD@Z)������ "public: __cdecl std::bad_any_cast::bad_any_cast(void)" (??0bad_any_cast@std@@QEAA@XZ) �������˸÷��� [E:\AMyDownload\universal_ble-main\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
universal_ble_filter_util.obj : error LNK2001: �޷��������ⲿ���� "protected: virtual void __cdecl std::bad_cast::_Doraise(void)const " (?_Doraise@bad_cast@std@@MEBAXXZ) [E:\AMyDownload\universal_ble-main\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
E:\AMyDownload\universal_ble-main\example\build\windows\x64\plugins\universal_ble\Debug\universal_ble_plugin.dll : fatal error LNK1120: 2 ���޷��������ⲿ���� [E:\AMyDownload\universal_ble-main\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
Error: Build process failed.

fotiDim, 2025-01-14:
@zjw78123sc
1) did you try flutter clean?
2) is the issue only with 0.14? Doe 0.13 example app work for you?
3) do other flutter projects build for you on the same machine?
4) did you try reinstalling Visual Studio?

zjw78123sc, 2025-01-14:
@fotiDim 
1.yes,have run flutter clean
2.0.13 also not work
3.what is this mean?I have many flutter project,but run one project in the same time.
4.you mean Visual Studio have some problem?

fotiDim, 2025-01-14:
>3.what is this mean?I have many flutter project,but run one project in the same time.

I am asking if other Flutter projects compile fine for you. Do you only have an issue with universal_ble?

I have:
- Windows 10.0.19045.5247
- Visual Studio Community 2022 17.8.16
- Flutter 3.27.1

[image]

I can run the example app just fine after flutter clean. I don't have Nuget installed so I doubt this is the issue.

Try maybe to reinstall Visual Studio and clone the project again in case anything has gone wrong.

zjw78123sc, 2025-01-15:
@fotiDim nice！update to the last version,project is fine

fotiDim, 2025-01-15:
@zjw78123sc nice, can we close this issue?

zjw78123sc, 2025-01-15:
@fotiDim OK

closed 2025-01-15 by fotiDim