Navideck/universal_ble #120: MacOS invalid range error

mskayali, 2025-01-02:
Swift/StringCharacterView.swift:163: Fatal error: String index is out of bounds

[image]

rohitsangwan01, 2025-01-08:
@mskayali do you have the UUID which is caushing this crash ?

oneplus1000, 2025-01-30:
I'm getting the same error. When receiving the service code "daf51c01"

[image]

rohitsangwan01, 2025-01-30:
@oneplus1000 @mskayali 
We have handled the UUID parsing on dart side now, so probably we will remove this formatting on native side, that should fix the issue, you can try to change line 351 to this and try again

```
services?.map { $0.uuidStr }
```

oneplus1000, 2025-01-30:
thank you  :smile:

fotiDim, 2025-02-15:
It should be addressed by https://github.com/Navideck/universal_ble/pull/138

closed 2025-03-17 by mskayali