Navideck/universal_ble #131: [IOS & MacOS] RequestMtu doesn't work like expected

Sravdar, 2025-01-17:
Tested and built on "Mac Mini m4", Flutter 3.27.1, xcode 16.2, universal ble 0.14.0
İphone 7, ios 15.8.3

On both of the devices i got the same behaviour.

On first RequestMtu call after boot and hot restarts it throws exception
> PlatformException(IllegalArgument, Unknown deviceId:<_TtC13universal_bleP33_7F14D50205415E5D4F8E2E2076A78ED716BleCentralDarwin: 0x15166b6e0>, null, null)

On following calls it doesn't throw exception but also doesn't set the mtu to target value and returns "23". Thing is peripheral device says mcu is set to its prefered value "495".

Sravdar, 2025-01-17:
I have found the cause. Calling 'requestMtu' before 'discoverServices' was causing this problem. 

But something to note is in example project i realized even though expected mtu is set to "247" i still get highest possible mtu. Not sure if this is intended or bug.

fotiDim, 2025-02-15:
> [@Sravdar](https://github.com/Sravdar) on few platforms like Apple or Windows, we can't negotiate MTU, its handled by platform itself, we can only get the Mtu value on these platforms, and yes discoverServices is required before getting Mtu, because its value depends on the services on few platforms

@rohitsangwan01 from what I understand it is not possible to set the MTU size starting [Android 14](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestMtu(int)). That only leaves Linux and older Android versions, right?

We should document all that to avoid confusion.

rohitsangwan01, 2025-02-16:
> [@rohitsangwan01](https://github.com/rohitsangwan01) from what I understand it is not possible to set the MTU size starting [Android [...]

Yes we should mention this in docs

> Also when calling 'requestMtu' before 'discoverServices' we should perhaps print a warning or throw an exception. The should apply to all methods that depend on services being already discovered. Another option would be to automatically call `discoverServices()` if not called already.

Yes we should test all platforms and see which one requires services for this , and throw proper exceptions

fotiDim, 2025-02-16:
> Yes we should mention this in docs
>
>
> Yes we should test all platforms and see which one requires services for this , and throw proper exceptions

@rohitsangwan01 Made a [PR](https://github.com/Navideck/universal_ble/pull/140) for the docs. I need your input for Linux.

For the services check we need another PR. Let's do it as part of this issue and include it in the upcoming release.

closed 2026-10-14 by fotiDim