Navideck/universal_ble #133: Error on windows

Bugnaldo, 2025-02-05:
Hi everyone,

sometimes on windows I receive this error:

[image]

I'm not sure but i think this happens during the scan process. If I manage to obtain more information about the error, I will let you know.

Bugnaldo, 2025-02-06:
It happened also today, and I'm pretty sure it was during scan process.  If it can helps, I can share you the two scan functions that I use in my apps:

`void _startScan(Function(DeviceModel device) callback, bool showBanned) {
    try {
      _listId.clear();

      UniversalBle.onScanResult = (event) {
        if (_filterName(event) &&
            !_listId.contains(event.deviceId) &&
            !_bannedDevice(event.name, showBanned)) {
          _listId.add(event.deviceId);
          if (kDebugMode) {
            log.debugPrint('Trovato dispositivo: $event');
          }
          DeviceModel deviceModel = DeviceModel.fromBleDevice(
              discoveredDevice: event,
              configStatus: _getState(event),
              kazByte: _getAdvertisingKazByte(event));
          callback(deviceModel);
        }
      };
      if (isScanning == false) {
        UniversalBle.startScan();
      }
      isScanning = true;
    } catch (e) {
      log.debugPrint('Error on startScan: $e');
      rethrow;
    }
  }

  Future<bool> _scanDeviceForFactory(
      {required String expectedDeviceSerial,
      required int durationInSeconds,
      required Function(DeviceModel device) callback,
      required bool isFirstScan}) async {
    try {
      bool isSuccess = false;
      String expectedDeviceName = 'KED_$expectedDeviceSerial';
      Completer<bool> completer = Completer<bool>();
      bool scanStopped = false;

      void stopScanSafely() {
        if (!scanStopped) {
          UniversalBle.stopScan();
          scanStopped = true;
        }
      }

      UniversalBle.onScanResult = (BleDevice device) {
        if (completer.isCompleted) {
          return; // Ignoro ulteriori risultati se il completer è già completato
        }

        try {
          if (device.name != null && device.name == expectedDeviceName) {
            log.debugPrint(
              "Dispositivo trovato: ${device.name}",
              logLevel: LogLevel.info,
            );

            stopScanSafely();

            // Valido i dati di advertising
            if (isFirstScan) {
              isSuccess = _validateAdvertisingData(device);
            } else {
              isSuccess = _secondAdvertisingValidation(device);
            }
            if (isSuccess) {
              //todo dichiarare fuori, con il secondo check sovrascrivere il devicemodel del primo check
              DeviceModel deviceModel = DeviceModel.fromBleDevice(
                discoveredDevice: device,
                configStatus: _getState(device),
                kazByte: _getAdvertisingKazByte(device),
              );

              try {
                callback(deviceModel);
              } catch (e) {
                log.debugPrint(
                  "Errore durante l'invocazione del callback: $e",
                  logLevel: LogLevel.error,
                );
              }
            }
            completer.complete(isSuccess);
          } else {
            log.debugPrint(
              "Nome dispositivo trovato: ${device.name}, atteso: $expectedDeviceName",
              logLevel: LogLevel.info,
            );
          }
        } catch (e) {
          log.debugPrint(
            "Errore durante la gestione del risultato della scansione: $e",
            logLevel: LogLevel.error,
          );
          if (!completer.isCompleted) {
            completer.complete(false);
          }
        }
      };

      try {
        // Avvia la scansione
        await UniversalBle.startScan(
          scanFilter: ScanFilter(
            withNamePrefix: [expectedDeviceName],
          ),
        );
        log.debugPrint("Scansione avviata...");

        // Tempo limite per la scansione
        Future.delayed(Duration(seconds: durationInSeconds)).then((_) {
          if (!completer.isCompleted) {
            log.debugPrint(
              "Tempo limite raggiunto senza trovare il dispositivo",
              logLevel: LogLevel.warning,
            );
            stopScanSafely();
            completer.complete(false);
          }
        });
      } catch (error) {
        log.debugPrint(
          "Errore durante l'avvio della scansione: $error",
          logLevel: LogLevel.error,
        );
        stopScanSafely();
        if (!completer.isCompleted) {
          completer.complete(false);
        }
      }

      return completer.future;
    } catch (e) {
      log.debugPrint('Error on scanDeviceForFactory: $e');
      rethrow;
    }
  }`

Bugnaldo, 2025-02-07:
I've seen that in my pubsec.yaml the version of Universal BLE was locked to 0.13.0 . I will try with new version

Bugnaldo, 2025-02-07:
I've had the same problem also with version 0.15.0, always while scanning for devices

rohitsangwan01, 2025-02-13:
@Bugnaldo are you able to replicate same on example app ?

nagatsuki, 2025-02-13:
The same problem occurred with universal_ble_example.

1. Run with "Start with debugging".
2. Press "Start Scan" button.
3. The Debug Assertion will occur after a few seconds to a few 10 seconds.

Stacktrace:
```
 	ntdll.dll!RtlpBreakPointHeap()	不明
 	ntdll.dll!RtlpValidateHeapEntry()	不明
 	ntdll.dll!RtlValidateHeap()	不明
 	KernelBase.dll!HeapValidate()	不明
>	ucrtbased.dll!_CrtIsValidHeapPointer(const void * block) 行 1407	C++
 	ucrtbased.dll!free_dbg_nolock(void * const block, const int block_use) 行 904	C++
 	ucrtbased.dll!_free_dbg(void * block, int block_use) 行 1030	C++
 	universal_ble_plugin.dll!operator delete(void * block) 行 34	C++
 	universal_ble_plugin.dll!operator delete(void * block, unsigned __int64 __formal) 行 31	C++
 	universal_ble_plugin.dll!std::_Deallocate<16>(void * _Ptr, unsigned __int64 _Bytes) 行 258	C++
 	universal_ble_plugin.dll!std::allocator<unsigned char>::deallocate(unsigned char * const _Ptr, const unsigned __int64 _Count) 行 949	C++
 	universal_ble_plugin.dll!std::vector<unsigned char,std::allocator<unsigned char>>::_Tidy() 行 2052	C++
 	universal_ble_plugin.dll!std::vector<unsigned char,std::allocator<unsigned char>>::~vector<unsigned char,std::allocator<unsigned char>>() 行 763	C++
 	universal_ble_plugin.dll!universal_ble::UniversalManufacturerData::~UniversalManufacturerData()	C++
 	universal_ble_plugin.dll!universal_ble::UniversalManufacturerData::`scalar deleting destructor'(unsigned int)	C++
 	universal_ble_plugin.dll!std::_Destroy_in_place<universal_ble::UniversalManufacturerData>(universal_ble::UniversalManufacturerData & _Obj) 行 293	C++
 	universal_ble_plugin.dll!std::_Any_small_RTTI::_Destroy_impl<universal_ble::UniversalManufacturerData>(void * const _Target) 行 81	C++
 	universal_ble_plugin.dll!std::any::reset() 行 206	C++
 	universal_ble_plugin.dll!std::any::_Assign(std::any _That) 行 293	C++
 	universal_ble_plugin.dll!std::any::operator=(const std::any & _That) 行 165	C++
 	universal_ble_plugin.dll!flutter::CustomEncodableValue::operator=(const flutter::CustomEncodableValue & __that)	C++
 	universal_ble_plugin.dll!std::_Variant_assign_visitor<std::monostate,bool,int,__int64,double,std::string,std::vector<unsigned char,std::allocator<unsigned char>>,std::vector<int,std::allocator<int>>,std::vector<__int64,std::allocator<__int64>>,std::vector<double,std::allocator<double>>,std::vector<flutter::EncodableValue,std::allocator<flutter::EncodableValue>>,std::map<flutter::EncodableValue,flutter::EncodableValue,std::less<flutter::EncodableValue>,std::allocator<std::pair<flutter::EncodableValue const ,flutter::EncodableValue>>>,flutter::CustomEncodableValue,std::vector<float,std::allocator<float>>>::operator()<flutter::CustomEncodableValue const &,12>(std::_Tagged<flutter::CustomEncodableValue const &,12> _Source) 行 706	C++
 	universal_ble_plugin.dll!std::_Variant_raw_visit1<2>::_Visit<std::_Variant_assign_visitor<std::monostate,bool,int,__int64,double,std::string,std::vector<unsigned char,std::allocator<unsigned char>>,std::vector<int,std::allocator<int>>,std::vector<__int64,std::allocator<__int64>>,std::vector<double,std::allocator<double>>,std::vector<flutter::EncodableValue,std::allocator<flutter::EncodableValue>>,std::map<flutter::EncodableValue,flutter::EncodableValue,std::less<flutter::EncodableValue>,std::allocator<std::pair<flutter::EncodableValue const ,flutter::EncodableValue>>>,flutter::CustomEncodableValue,std::vector<float,std::allocator<float>>>,std::_Variant_storage_<0,std::monostate,bool,int,__int64,double,std::string,std::vector<unsigned char,std::allocator<unsigned char>>,std::vector<int,std::allocator<int>>,std::vector<__int64,std::allocator<__int64>>,std::vector<double,std::allocator<double>>,std::vector<flutter::EncodableValue,std::allocator<flutter::EncodableValue>>,std::map<flutter::EncodableValue,flutter::EncodableValue,std::less<flutter::EncodableValue>,std::allocator<std::pair<flutter::EncodableValue const ,flutter::EncodableValue>>>,flutter::CustomEncodableValue,std::vector<float,std::allocator<float>>> const &>(unsigned __int64 _Idx, std::_Variant_assign_visitor<std::monostate,bool,int,__int64,double,std::string,std::vector<unsigned char,std::allocator<unsigned char>>,std::vector<int,std::allocator<int>>,std::vector<__int64,std::allocator<__int64>>,std::vector<double,std::allocator<double>>,std::vector<flutter::EncodableValue,std::allocator<flutter::EncodableValue>>,std::map<flutter::EncodableValue,flutter::EncodableValue,std::less<flutter::EncodableValue>,std::allocator<std::pair<flutter::EncodableValue const ,flutter::EncodableValue>>>,flutter::CustomEncodableValue,std::vector<float,std::allocator<float>>> && _Func, const std::_Variant_storage_<0,std::monostate,bool,int,__int64,double,std::string,std::vector<unsigned char,std::allocator<unsigned char>>,std::vector<int,std::allocator<int>>,std::vector<__int64,std::allocator<__int64>>,std::vector<double,std::allocator<double>>,std::vector<flutter::EncodableValue,std::allocator<flutter::EncodableValue>>,std::map<flutter::EncodableValue,flutter::EncodableValue,std::less<flutter::EncodableValue>,std::allocator<std::pair<flutter::EncodableValue const ,flutter::EncodableValue>>>,flutter::CustomEncodableValue,std::vector<float,std::allocator<float>>> & _Obj) 行 621	C++
 	universal_ble_plugin.dll!std::_Variant_raw_visit<std::_Variant_storage_<0,std::monostate,bool,int,__int64,double,std::string,std::vector<unsigned char,std::allocator<unsigned char>>,std::vector<int,std::allocator<int>>,std::vector<__int64,s [... cut ...] em::RaiseEvent(void)	不明
 	Windows.Devices.Bluetooth.dll!wil::details::AsyncEventInvocationPolicy<1>::HandleAsyncEventInvoke(class wil::srwlock &,class wil::details::AsyncEventWorkQueue &)	不明
 	Windows.Devices.Bluetooth.dll!wil::details::EventInvocationContext<1>::AsyncEventWorkCallback(struct _TP_CALLBACK_INSTANCE *,void *,struct _TP_WORK *)	不明
 	ntdll.dll!TppWorkpExecuteCallback()	不明
 	ntdll.dll!TppWorkerThread()	不明
 	kernel32.dll!BaseThreadInitThunk()	不明
 	ntdll.dll!RtlUserThreadStart()	不明
```

Bugnaldo, 2025-02-13:
@rohitsangwan01 The same issue occurs with the example as well. Try starting the scan, then stopping it, and then restarting the scan

fotiDim, 2025-02-21:
@Bugnaldo @nagatsuki
We will need more info to be able to track this down. We tried on 4 different machines (Windows 10 and 11) and all of them work fine for us.

1. Please use [Bluetooth LE Explorer](https://apps.microsoft.com/detail/9n0ztkf1qd98) and see if you are getting similar issues
2. Provide your `flutter doctor` output
3. Describe your machine's hardware in detail: brand, model, BT dongle (if external)
4. Try running the example app through Visual Studio. Enable Visual **Studio's AddressSanitizer** to detect out-of-bounds writes. Also, set environment variables to help track memory issues:`set _CRTDBG_REPORT_FLAG = _CRTDBG_ALLOC_MEM_DF | _CRTDBG_LEAK_CHECK_DF`.

Bugnaldo, 2025-03-04:
@rohitsangwan01 I've tried with #143 but the issue is still present

@fotiDim 
1) With [Bluetooth LE Explorer](https://apps.microsoft.com/detail/9n0ztkf1qd98) i don't have issues
2) My flutter doctor output is: C:\dev\repos\KedAppInstallerFlutter git:[main]
flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, 3.29.0, on Microsoft Windows [Versione 10.0.26100.3194], locale it-IT)
[√] Windows Version (11 Pro 64-bit, 24H2, 2009)
[√] Android toolchain - develop for Android devices (Android SDK version 35.0.0)
[√] Chrome - develop for the web
[√] Visual Studio - develop Windows apps (Visual Studio Community 2022 17.12.1)
[√] Android Studio (version 2024.2)
[√] VS Code (version 1.97.2)
[√] Connected device (3 available)
[√] Network resources

• No issues found!
3) My machine is a Dell latitude 7640, system language italian, no external dongles (pc has integrated Intel® Wi-Fi 6E AX211 160MHz), 
4) I will try with it and update this message

fotiDim, 2025-03-08:
I managed to reproduce the issue once. It took a lot of attempts and patience and I am not sure of the exact steps that triggered it. It could also be related to a specific scanned device.

@Bugnaldo does it happen frequently and consistently for you?

fotiDim, 2025-03-11:
@Bugnaldo @nagatsuki @Silvesterrr could you please try with https://github.com/Navideck/universal_ble/pull/146?

Bugnaldo, 2025-03-17:
I've tried and everything is working fine, I will let you know if something happens in next days. Thank you for your work

fotiDim, 2025-03-17:
@Bugnaldo great! We are planning a few more Windows improvements. We will include them all in the next release so better wait for that.

Kudos to @rohitsangwan01 for the fix!

closed 2025-03-18 by rohitsangwan01

fotiDim, 2025-03-19:
@Bugnaldo please test once again with 0.17.0 which just got released.