Navideck/universal_ble #135: UniversalBle.OnValueChange returns synchronously when UniversalBle.setNotifiable multiple

zjw78123sc, 2025-02-13:
I UniversalBle.setNotifiable two characteristic (eg. A and B) , A is 128 data/ms , B is command data,
when A return data continuously, B return a command data, I find it will return B after A return All the data.
Can   UniversalBle.OnValueChange Separate listening A and B？

rohitsangwan01, 2025-02-13:
@zjw78123sc Which platform you are using ?, and Events should be received parallely for all subscribed characteristics
Unless the platform or your peripheral itself sending them synchronously,
It would be great to have logs with: What your are getting ( From UniversalBle) and Whats expected ( From another App maybe )

zjw78123sc, 2025-02-13:
Windows Platform

rohitsangwan01, 2025-02-21:
@zjw78123sc i tried running similar case on Windows, 
Char A continuously sending updated
Char B sending updates with some delay
And both updates are captured instantly and parallelly, so probably your peripheral might be sending the updates synchronously

fotiDim, 2025-05-12:
@zjw78123sc did you manage to resolve this?

fotiDim, 2026-10-15:
Closing due to lack of feedback. We can reopen if the issue is still there.

closed 2026-10-15 by fotiDim