Navideck/universal_ble #136: Failed to subscribe via UniversalBle.setNotifiable on iOS"

lemos1235, 2025-02-14:
when i execute
```
UniversalBle.setNotifiable(deviceId, "0000180d-0000-1000-8000-00805f9b34fb",
                      "00002a2b-0000-1000-8000-00805f9b34fb", BleInputProperty.indication);
```

I got error message:

```
flutter: [BleService{uuid: e73e0001-ef1b-4e74-8291-2e4f3164f3b5, characteristics: [BleCharacteristic{uuid: e73e0002-ef1b-4e74-8291-2e4f3164f3b5, properties: [read]}]}, BleService{uuid: 0000180d-0000-1000-8000-00805f9b34fb, characteristics: [BleCharacteristic{uuid: 00002a2b-0000-1000-8000-00805f9b34fb, properties: [read, notify, indicate]}]}]
[ERROR:flutter/runtime/dart_vm_initializer.cc(40)] Unhandled Exception: PlatformException(10, The attribute could not be found., null, null)
#0      UniversalBlePlatformChannel.setNotifiable (package:universal_ble/src/universal_ble_pigeon/universal_ble.g.dart:462:7)
<asynchronous suspension>
#1      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:1004:15)
<asynchronous suspension>
#2      UniversalBle.setNotifiable (package:universal_ble/src/universal_ble.dart:158:12)
<asynchronous suspension>
```

Additionally, the same error occurs in the sample app.

closed 2025-02-15 by lemos1235

Pertempto, 2025-03-12:
@lemos1235  did you find a fix for this?

Pertempto, 2025-03-12:
It turned out this error in our situation was caused by our device not being correctly configured with the BLE2902 descriptor.

fotiDim, 2025-03-13:
@Pertempto on which platform did you encounter the issue? Was it iOS? What kind of error would you expect?