Navideck/universal_ble #145: auto-connect support

bbilger, 2025-03-08:
Any chance to add support for auto-connect - for Android and iOS? Could imagine that to be a new parameter on the connect-method.

- https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback)
- https://developer.apple.com/documentation/corebluetooth/cbconnectperipheraloptionenableautoreconnect

fotiDim, 2025-03-08:
@bbilger Can you also please have a look regarding support in Windows, Linux and web and add that to the issue?

bbilger, 2025-03-08:
AFAIK there's no such thing for Web, Linux and Windows. So that parameter would be ignored on those platforms.

quasarsuport-del, 2026-01-07:
I would like to join the request to add support for automatic reconnection on Android and iOS.

In practical BLE use cases—especially mobile and embedded—disconnections are frequent and outside the application’s control. The lack of built-in auto-reconnect significantly complicates application logic and leads to duplicated solutions across projects.

This feature would be very important to me and would greatly improve the library’s usability in production scenarios.

Thank you for considering this request.