Navideck/universal_ble #149: isPaired status is not correct in Apple devices

aavaa-farnood, 2025-03-19:
Just try your own example with an Apple device, all `isPaired` status are `null` value.

I tried it on: 
```
MacBook Air M2
macOS 15.3.2 (24D81)
```

and 

```
iPhone 13
iOS 18.3.1
```

aavaa-farnood, 2025-03-21:
@fotiDim Thank you for the explanation and the update in the docs.

I looked more into it, but I don't seem to find any docs on how to define or find those encrypted read/write chars. Also in the example project, there is no use of `BleCommand` to learn how to use it.

Could you please guide me on this?

Thanks.

aavaa-farnood, 2025-03-21:
@fotiDim 

Also I noticed this part is wrong in the docs:

```
bool? isPaired = await UniversalBle.pair(deviceId);
```

The `.pair()` method returns `void` and it does not throw any errors either on failing to connect on macOS unlike Windows.

fotiDim, 2025-03-22:
@aavaa-farnood you are right about the wrong part in the docs. That was a leftover after a recent architectural change we did.

I have updated the readme once again. I hope it is clearer now. You might find the `Discovering encrypted characteristic` section useful. Please let me know if you have any questions after this. I would like to make it as clear  as possible.

Note that we really cannot use `BleCommand` in the example app as this is device-specific.

closed 2025-05-12 by fotiDim

fotiDim, 2026-03-05:
@RageshAntony is the device paired? If yes, you should be getting `true`.

RageshAntony, 2026-03-10:
@fotiDim 
Okay, Got it. 

And, does `Universal_ble` support pairing via '`AccessorySetupKit`' ?