Navideck/universal_ble #156: Gatt timeout

spiky-star, 2025-05-06:
I have a problem, why it happened

`gatt_connect_cback: remove untrusted client status, bda=xx:xx:xx:xx:c4:aa
MF73274  05-05 20:58:51.682 21386 21421 I bluetooth: system/stack/gap/gap_ble.cc:306 client_connect_cback: No active GAP service found for peer:xx:xx:xx:xx:c4:aa callback:Disconnected
MF73275  05-05 20:58:51.682 21386 21421 I bt_bta_gattc: system/bta/gatt/bta_gattc_act.cc:1418 bta_gattc_conn_cback: Disconnected att_id:3 addr:xx:xx:xx:xx:c4:aa, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
MF73276  05-05 20:58:51.682 21386 21421 I bt_bta_gattc: system/bta/gatt/bta_gattc_act.cc:1418 bta_gattc_conn_cback: Disconnected att_id:4 addr:xx:xx:xx:xx:c4:aa, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
MF73277  05-05 20:58:51.682 21386 21421 I bt_bta_gattc: system/bta/gatt/bta_gattc_act.cc:1418 bta_gattc_conn_cback: Disconnected att_id:5 addr:xx:xx:xx:xx:c4:aa, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
MF73278  05-05 20:58:51.682 21386 21421 I bt_bta_gattc: system/bta/gatt/bta_gattc_act.cc:1418 bta_gattc_conn_cback: Disconnected att_id:6 addr:xx:xx:xx:xx:c4:aa, transport:BT_TRANSPORT_LE reason:GATT_CONN_TIMEOUT
MF73279  05-05 20:58:51.682 21386 21421 W bt_btm_sec: system/stack/btm/btm_sec.cc:3933 btm_sec_disconnected: Got uncommon disconnection reason:HCI_ERR_CONNECTION_TOUT handle:0x0010 comment:stack::acl::btm_acl::btm_acl_disconnected`

fotiDim, 2025-05-12:
@spiky-star could you provide some more details regarding your setup? Did you try using the example app?

spiky-star, 2025-05-18:
It happened at long time background

closed 2026-03-13 by fotiDim