Navideck/universal_ble #158: Build issue on windows

alextekartik, 2025-05-13:
Using current flutter 3.29.3 stable
I cannot manage to run the example on windows as i get the following error:

```
LINK : warning LNK4217: symbol '??1bad_cast@std@@UEAA@XZ (public: virtual __cdecl std::bad_cast::~bad_cast(void))' defined in 'universal_ble.g.obj' is imported by 'universal_ble_filter_util.obj' in function '"public: virtual __cdecl std::bad_any_cast::~bad_any_cast(void)" (??1bad_any_cast@std@@UEAA@XZ)' [C:\opt\extdevx\git\github.com\Navideck\universal_ble\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
universal_ble_filter_util.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl std::bad_cast::bad_cast(char const *)" (__imp_??0bad_cast@std@@QEAA@PEBD@Z) referenced in function "public: __cdecl std::bad_any_cast::bad_any_cast(void)" (??0bad_any_cast@std@@QEAA@XZ) [C:\opt\extdevx\git\github.com\Navideck\universal_ble\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
universal_ble_filter_util.obj : error LNK2001: unresolved external symbol "protected: virtual void __cdecl std::bad_cast::_Doraise(void)const " (?_Doraise@bad_cast@std@@MEBAXXZ) [C:\opt\extdevx\git\github.com\Navideck\universal_ble\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
C:\opt\extdevx\git\github.com\Navideck\universal_ble\example\build\windows\x64\plugins\universal_ble\Debug\universal_ble_plugin.dll : fatal error LNK1120: 2 unresolved externals [C:\opt\extdevx\git\github.com\Navideck\universal_ble\example\build\windows\x64\plugins\universal_ble\universal_ble_plugin.vcxproj]
Error: Build process failed.
```

Is there any step additionnal step required on windows ?

Thank you!

fotiDim, 2025-05-13:
@alextekartik make sure to update Visual Studio to the[ latest version](https://visualstudio.microsoft.com/). Last time, I had to re-install mine from scratch because Visual Studio Installer did not report any updates.

alextekartik, 2025-05-13:
@fotiDim That was it thank you! Although flutter doctor did not report any error. Sorry for the noise.