Navideck/universal_ble #159: 部分windows上 在调用 await UniversalBle.startScan(); 后发生崩溃

jiangchengze, 2025-05-21:
错误模块名称: universal_ble_plugin.dll，版本: 0.0.0.0，时间戳: 0x682d92da
异常代码: 0xc0000005
错误偏移量: 0x000000000000e002
错误进程 ID: 0x2c54
错误应用程序启动时间: 0x01dbca391e7b4f45
错误应用程序路径: C:\Program Files\LabelExpert\labelexpert.exe
错误模块路径: C:\Program Files\LabelExpert\universal_ble_plugin.dll
报告 ID: 1e1052c2-5807-4fa3-b7b1-5ad1e7f7731b
错误程序包全名: 
错误程序包相对应用程序 ID:

rohitsangwan01, 2025-05-26:
@jiangchengze can you provide some info or share your code to start scanning
is example app working fine for you ?

rohitsangwan01, 2025-05-26:
Are you trying to compile app and run it on another computer ?
if thats the case, is it working fine in your development computer ?

jiangchengze, 2025-09-02:
I have tested this: the compilation and operation work normally on my own computer. However, when other users use it with a Bluetooth adapter plugged in, calling await UniversalBle.startScan(); causes a crash. When the Bluetooth adapter is not plugged in, calling await UniversalBle.startScan(); does not result in a crash. Currently, the version of universal_ble being used is ^0.21.0.

fotiDim, 2025-10-03:
@jonasbark have you confirmed that your issue happens on `startScan()`?

@jiangchengze do you have details of the problematic computer? Bluetooth dongle? OS version? We will need that to be able reproduce it. None of our machines has caused a similar issue.

jonasbark, 2025-11-15:
doesn't crash on my virtual machine, so I can't confirm it.

fotiDim, 2026-10-15:
We have done a ton of stability improvements since. Closing and we can reopen if the issue is still there.

closed 2026-10-15 by fotiDim