Navideck/universal_ble #167: `BleCharacteristic.notifications.listen` and `BleCharacteristic.onValueReceived` not receiving data.

ljmatan, 2025-07-05:
First of all, this looks like really nice work and helped me a lot. 

With iPad Pro M4 13 inch, iOS `18.5`, Flutter version `3.33.0-1.0.pre.826`, I'm having issues where the methods `BleCharacteristic.notifications.listen` and `BleCharacteristic.onValueReceived` are not being triggered.

Example code:

```dart
    try {
      await _scanCharacteristic!.notifications.subscribe();
      print('Subscribed to characteristic notifications.');
      _scanCharacteristic!.notifications.listen(
        (value) {
          print('Received indication: $value');
        },
        onDone: () {
          print('DONE');
        },
        onError: (error) {
          print('ERROR: $error');
        },
      );
    } catch (e) {
      print('Error subscribing to notifications: $e');
    }
    _scanCharacteristic!.onValueReceived.listen(
      (value) {
        print('Data received: $value');
      },
      onDone: () {
        print('DONE');
      },
      onError: (error) {
        print('ERROR: $error');
      },
    );
    uble.UniversalBle.onValueChange = (
      String deviceId,
      String characteristicId,
      Uint8List value,
    ) {
      print(
        utf8.decode(value),
      );
    };
```

The values are however received via `uble.UniversalBle.onValueChange`, so the library does work after all. The `example` app helped me with working out the issue.

rohitsangwan01, 2025-07-06:
Thats strange, can you please try `BleCharacteristic.notifications.listen` and `BleCharacteristic.onValueReceived` with pr https://github.com/Navideck/universal_ble/pull/168

ljmatan, 2025-07-06:
I'm only trying to subscribe to one characteristic: 

```
    final _scanCharacteristicUuid = '00002aa1-0000-1000-8000-00805f9b34fb';
    _services = await _barcodeScanner!.discoverServices();
    for (final service in _services!) {
      if (service.uuid == _scanServiceUuid) {
        _scanCharacteristic = service.characteristics.firstWhereOrNull(
          (characteristic) {
            return characteristic.uuid == _scanCharacteristicUuid;
          },
        );
      }
    }
    if (_scanCharacteristic == null) {
      throw Exception(
        'BLE service $_scanServiceUuid characteristic $_scanCharacteristicUuid not found.',
      );
    }
    await _scanCharacteristic!.notifications.subscribe();
```

I've also confirmed this is what's printed in the console with the `UniversalBle.onValueChange` method:

[image]

ljmatan, 2025-07-06:
> How many events are you getting in onValueChange ?

I'm using a Birch BD-3510 barcode scanner, I get events on every successful QR code scan with this device.

> Are you getting event right after subscribe ?

Yes, or perhaps the above answers it.

> Maybe try to set listener of onValueReceived before calling subscribe method

Unfortunately, that is also not working:

```
    _scanCharacteristic!.notifications.listen(
      (value) {
        print('Received indication: $value');
      },
      onDone: () {
        print('DONE');
      },
      onError: (error) {
        print('ERROR: $error');
      },
    );
    _scanCharacteristic!.onValueReceived.listen(
      (value) {
        print('Data received: $value');
      },
      onDone: () {
        print('DONE');
      },
      onError: (error) {
        print('ERROR: $error');
      },
    );
    await _scanCharacteristic!.notifications.subscribe();
```

This outputs nothing to the console.

ljmatan, 2025-07-06:
Perhaps it's also worth mentioning that I've tried with `indications` as well with the same result, not only with `notifications`.

ljmatan, 2025-07-06:
Thank you for your work and support as well.

I've downloaded the nRF Connect Android app and it appeared to be fully working:

[log.txt](https://github.com/user-attachments/files/21090040/log.txt)

Also, after changing some other dependencies and running `flutter pub get`, the notification listener appears to be fully working as well: 

[image]

I do receive indications as well even if I didn't invoke the `inidications.subscribe` method however. I also appear to receive more inputs from the `listen` methods than from the `UniversalBle.onValueChange` method, although that's not in the scope of this ticket.

I do apologise for taking up your time! Looks like I didn't actually update the dependency as necessary. 

If I can help you out with further testing, please feel free to reopen the issue, and thanks again.

ljmatan, 2025-07-07:
My apologies.

I have executed the following command into the terminal:

`flutter clean && flutter packages upgrade && flutter packages get`

These are my `pubspec.yaml` contents:

```
  universal_ble:
    git:
      url: https://github.com/Navideck/universal_ble
      ref: main
```

Having then done the test again, the listener is not working: 

https://github.com/user-attachments/assets/4f68c6ac-5f52-4c6e-9d4d-2efccce6a4b2

It seems to only be working with the `improve-characteristic-value-listener` branch.