Navideck/universal_ble #170: Unable to get BleCharacteristic.onValueReceived to work

tadelv, 2025-07-07:
Hi, 

I am seeing the same issue as reported here: https://github.com/Navideck/universal_ble/issues/167

For me, only `UniversalBle.onValueChange` is working, nothing else.

here is the sample code (that isn't working):
```
    final characteristic = await _device.getCharacteristic(dataUUID,
        service: serviceUUID, preferCached: false);
    characteristic.onValueReceived.listen(_parseNotification, onError: (er) {
      print("error: ${er}");
    }, cancelOnError: true);
    print("characteristic: ${characteristic}");
    await characteristic.notifications.subscribe();
```

and my pubspec.yaml:
```
  universal_ble: 
    git:
      url: https://github.com/Navideck/universal_ble.git
      ref: improve-characteristic-value-listener
```

rohitsangwan01, 2025-07-07:
@tadelv can you try to update flutter dependencies to ensure you are on ‘ improve-characteristic-value-listener’ branch, also cross check in pubspec.lock file

tadelv, 2025-07-07:
@rohitsangwan01 how? :D 
indeed, after running `flutter pub upgrade` I got it to work. Here's my pubspec.lock for reference:
```
  universal_ble:
    dependency: "direct main"
    description:
      path: "."
      ref: improve-characteristic-value-listener
      resolved-ref: "7b20b0bc75e723357be9e997bfb2c3a889f83853"
      url: "https://github.com/Navideck/universal_ble.git"
```

Thank you so much!

closed 2025-07-08 by rohitsangwan01