Navideck/universal_ble #180: startScan() fails in release builds with channel-error

endmr11, 2025-09-30:
The startScan() method works perfectly in debug builds but fails in release builds with the following error:

PlatformException(channel-error, Unable to establish connection on channel: "dev.flutter.pigeon.universal_ble.UniversalBlePlatformChannel.startScan"., null, null)

Environment

universal_ble version: [0.21.0]
Flutter version: [3.9.0(Dart), 3.35.2(Flutter)]
Platform: Android
Build configuration: Release with minifyEnabled = true

Steps to Reproduce

Run the app in release mode: flutter run apk --release
Install the APK on device
Call UniversalBle.startScan()
Error occurs

startScan() should work in both debug and release builds.

Works in debug builds, fails in release builds with channel connection error.

The issue appears when building with code minification/obfuscation enabled. Debug builds work fine.

endmr11, 2025-10-01:
@fotiDim

fotiDim, 2025-10-03:
@endmr11 `flutter run apk --release` gives `Target file "apk" not found`. Did you mean `flutter run --release`?
I tested it with the example with
```
shrinkResources true
minifyEnabled true
```
and worked fine for me. If you are using Proguard you need to adjust your rules accordingly.

endmr11, 2025-10-05:
@fotiDim  I uninstalled progurad, shrinkResources, and others. Could the problem be related to AGP? My current AGP version is 8.7.0

fotiDim, 2025-10-05:
@endmr11 Can you please test with the example app?

fotiDim, 2025-10-05:
@endmr11 you need to crosscheck the differences between the 2 projects. Can you provide a minimal sample project where the issue is reproducible?

endmr11, 2025-10-05:
I compared the two projects completely, and everything was the same, but it still didn't work. In your universal_ble.g.dart file, this code returns null.

final List<Object?>? pigeonVar_replyList =
await pigeonVar_channel.send(<Object?>[filter]) as List<Object?>?;

endmr11, 2025-10-05:
I think there was a conflict with another subproject. I don't know which one, but I solved it :)

closed 2025-10-05 by endmr11